project-chip/connectedhomeip #31672: M5 board keeps resetting itself after pairing to Home app and the accessory goes to No Response state

phonnakasturi-apple, 2024-01-25:
### Reproduction steps

With the latest SHA, M5 board keeps resetting itself after pairing to Home app and the accessory goes to No Response state

SHA: 7f5e208f78847466816cd0544617ab077d3d0bb5

1. cd ~/connectedhomeip
2. git reset --hard 7f5e208f78847466816cd0544617ab077d3d0bb5
3. source scripts/bootstrap.sh
4. source scripts/activate.sh
5. cd ~/tools/esp-idf 
6. git checkout v5.1
7. ./install.sh
8. . ./export.sh
9. cd ~/connectedhomeip/examples/all-clusters-app/esp32
10. export ZAP_INSTALL_PATH=~/Downloads/zap-mac/
11. idf.py build
12. idf.py menuconfig -> select M5 from Demo
13. idf.py -D 'SDKCONFIG_DEFAULTS=sdkconfig_m5stack.defaults' erase-flash build flash monitor
14. Launch HomeApp
15. Pair M5 board with HomeApp

Results:
Soon after pairing, M5 board keeps resetting itself and the accessory goes to No Response state

### Bug prevalence

all the time

### GitHub hash of the SDK that was being used

7f5e208f78847466816cd0544617ab077d3d0bb5

### Platform

darwin

### Type

Core SDK Crash

phonnakasturi-apple, 2024-01-25:
Attaching the logs: [M5_logs.txt](https://github.com/project-chip/connectedhomeip/files/14045508/M5_logs.txt)

**Crash message:**
```
abort() was called at PC 0x401056d2 on core 1
0x401056d2: GetEpochTS(unsigned long&) at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/examples/energy-management-app/energy-management-common/src/EnergyEvseDelegateImpl.cpp:1477 (discriminator 9)

Backtrace: 0x40081ac2:0x3ffe9350 0x40093fa5:0x3ffe9370 0x4009873a:0x3ffe9390 0x401056d2:0x3ffe9400 0x40105719:0x3ffe9430 0x40105758:0x3ffe9460 0x400e16fa:0x3ffe9480 0x400dbdb2:0x3ffe94b0 0x400dd175:0x3ffe9550 0x400dd288:0x3ffe9570 0x400dd4cf:0x3ffe9650 0x400dd6e5:0x3ffe9770 0x400dd750:0x3ffe97b0 0x40145d65:0x3ffe97d0 0x40145d75:0x3ffe97f0 0x4014e64d:0x3ffe9810 0x4014e970:0x3ffe9830 0x4014e9a5:0x3ffe98a0 0x4009446a:0x3ffe98c0
0x40081ac2: panic_abort at /Users/priyankahonnakasturi/tools/esp-idf/components/esp_system/panic.c:452

0x40093fa5: esp_system_abort at /Users/priyankahonnakasturi/tools/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x4009873a: abort at /Users/priyankahonnakasturi/tools/esp-idf/components/newlib/abort.c:38

0x401056d2: GetEpochTS(unsigned long&) at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/examples/energy-management-app/energy-management-common/src/EnergyEvseDelegateImpl.cpp:1477 (discriminator 9)

0x40105719: chip::app::Clusters::EnergyEvse::EvseSession::RecalculateSessionDuration() at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/examples/energy-management-app/energy-management-common/src/EnergyEvseDelegateImpl.cpp:1554

0x40105758: chip::app::Clusters::EnergyEvse::EnergyEvseDelegate::GetSessionDuration() at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/examples/energy-management-app/energy-management-common/src/EnergyEvseDelegateImpl.cpp:1448

0x400e16fa: chip::app::Clusters::EnergyEvse::Instance::Read(chip::app::ConcreteReadAttributePath const&, chip::app::AttributeValueEncoder&) at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/app/clusters/energy-evse-server/energy-evse-server.cpp:81

0x400dbdb2: chip::app::ReadSingleClusterData(chip::Access::SubjectDescriptor const&, bool, chip::app::ConcreteReadAttributePath const&, chip::app::AttributeReportIBs::Builder&, chip::app::AttributeValueEncoder::AttributeEncodeState*) at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/app/util/ember-compatibility-functions.cpp:456
 (inlined by) chip::app::ReadSingleClusterData(chip::Access::SubjectDescriptor const&, bool, chip::app::ConcreteReadAttributePath const&, chip::app::AttributeReportIBs::Builder&, chip::app::AttributeValueEncoder::AttributeEncodeState*) at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/app/util/ember-compatibility-functions.cpp:590

0x400dd175: chip::app::reporting::Engine::RetrieveClusterData(chip::Access::SubjectDescriptor const&, bool, chip::app::AttributeReportIBs::Builder&, chip::app::ConcreteReadAttributePath const&, chip::app::AttributeValueEncoder::AttributeEncodeState*) at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/app/reporting/Engine.cpp:86

0x400dd288: chip::app::reporting::Engine::BuildSingleReportDataAttributeReportIBs(chip::app::ReportDataMessage::Builder&, chip::app::ReadHandler*, bool*, bool*) at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/app/reporting/Engine.cpp:193

0x400dd4cf: chip::app::reporting::Engine::BuildAndSendSingleReportData(chip::app::ReadHandler*) at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/app/reporting/Engine.cpp:536 (discriminator 3)

0x400dd6e5: chip::app::reporting::Engine::Run() at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/app/reporting/Engine.cpp:657 (discriminator 3)

0x400dd750: chip::app::reporting::Engine::Run(chip::System::Layer*, void*) at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/third_party/connectedhomeip/src/app/reporting/Engine.cpp:609

0x40145d65: chip::System::TimerData::Callback::Invoke() const at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/system/SystemTimer.h:61
 (inlined by) chip::System::TimerPool<chip::System::TimerList::Node>::Invoke(chip::System::TimerList::Node*) at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/system/SystemTimer.h:237

0x40145d75: chip::LambdaBridge::Initialize<chip::System::LayerImplFreeRTOS::ScheduleWork(void (*)(chip::System::Layer*, void*), void*)::{lambda()#1}>(chip::System::LayerImplFreeRTOS::ScheduleWork(void (*)(chip::System::Layer*, void*), void*)::{lambda()#1} const&)::{lambda(std::aligned_storage<24u, 4u>::type const&)#1}::_FUN(std::aligned_storage<24u, 4u>::type const&) at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../.. [... cut ...]
 (inlined by) chip::DeviceLayer::Internal::GenericPlatformManagerImpl_FreeRTOS<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(void*) at /Users/priyankahonnakasturi/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_FreeRTOS.ipp:280

0x4009446a: vPortTaskWrapper at /Users/priyankahonnakasturi/tools/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

```

bzbarsky-apple, 2024-01-25:
@jamesharrow can you take a look?  This is crashing in EVSE code....

jamesharrow, 2024-01-25:
Sure.

At first sight it looks like it’s either failing to get timesync and the platform isn’t handling that well, or it’s related to persistent attributes.

There was a fix put in yesterday to the example energy management app esp32 which ensures the persistence layer was initialised earlier.

In order to understand what’s happening better can we get more logs on what the HomeApp is trying to do? Which HomeApp is this? It is something that is public so we can use to reproduce it?

jamesharrow, 2024-01-25:
Ok the issue (having looked at the code) is that this platform doesn't support GetClock_RealTimeMS();
So when instantiating this  it calls:
    /* If the GetClock_RealTimeMS returns CHIP_ERROR_UNSUPPORTED_CHIP_FEATURE, then
     * This platform cannot ever report real time !
     * This should not be certifiable since getting time is a Mandatory
     * feature of EVSE Cluster
     */
    VerifyOrDie(err != CHIP_ERROR_UNSUPPORTED_CHIP_FEATURE);

So the app is deliberately die'ing to expose this problem.

The challenge here is that all-clusters-app needs to support all platforms which may or may not support GetRealTimeMS() whereas in practise you won't be able to certify EVSE on platforms that don't implement this capability.

@bzbarsky-apple what do you suggest here?
I can add a more graceful handling to get around this - i.e. remove the VerifyOrDie() - then it will return the error up to the callee which will again mean that some of the EVSE features like sessions and automatic session expiry etc won't work properly.

Alternatively we can ask silicon providers to try to implement GetRealTimeMS()?

bzbarsky-apple, 2024-01-25:
@jamesharrow Fundamentally the options seem to be:

1. Not include EVSE in all-clusters-app if we have some other example app that can be used for testing it, and make sure that other app only runs on platforms that have a real-time clock.
2. Have the all-clusters-app on EVSE fail more gracefully, so that things that are not trying to do anything EVSE-specific (e.g. trying to do a wildcard read/subscribe, which is what's going on here) don't end up killing the server process.  Right now this behavior basically blocks all-clusters-app on ESP32 from being used for _anything_ that involves testing wildcarding behavior.

jamesharrow, 2024-01-26:
Sounds like option 2 would be best - I think it just needs the VerifyOrDie removing - that was a PR comment from someone I think to add it.

I think TC_EEVSE_2_2 will catch this error since it checks the enable charging times out at a specific real time (+X seconds into the future)

PR #32058 (2024-02-16): Fixes: #31672 avoids a deliberate crash on targets which don't suppor…
Fixes: #31672 
 - #31672 

This was caused by a VerifyOrDie() when the platform System::SystemClock().GetClock_RealTimeMS(cTMs) returns CHIP_ERROR_UNSUPPORTED_CHIP_FEATURE.

- [x] Removed the VerifyOrDie and made it fail more gracefully

Note platforms not supporting realtime clock cannot support EVSE cluster.
commit 5dfaed47 (2024-02-16): Fixes: #31672 avoids a deliberate crash on targets which don't suppor… (#32058)

* Fixes: #31672 avoids a deliberate crash on targets which don't support real-time clock

* Added clearer error message about checking EVSE certification requirements.
closed 2024-02-16 by mergify[bot]