project-chip/connectedhomeip #31680: Nimble SPP Server example not working

riegaz, 2024-01-25:
### Reproduction steps

- Use a ESP32 (chip v1.0)
- Use ESP-IDF 5.1.2
- Run the nimble SPP server example
- The ESP32 gets stuck at advertising

### Bug prevalence

always

### GitHub hash of the SDK that was being used

5.1.2

### Platform

esp32

### Anything else?

Hi, this is the log:

I (31) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (31) boot: compile time Jan 25 2024 11:34:45
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v1.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=190a0h (102560) map
I (141) esp_image: segment 1: paddr=000290c8 vaddr=3ffbdb60 size=049b8h ( 18872) load
I (149) esp_image: segment 2: paddr=0002da88 vaddr=40080000 size=02590h (  9616) load
I (153) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=61f38h (401208) map
I (300) esp_image: segment 4: paddr=00091f60 vaddr=40082590 size=17ec0h ( 97984) load
I (355) boot: Loaded app from partition at offset 0x10000
I (355) boot: Disabling RNG early entropy source...
I (366) cpu_start: Multicore app
I (367) cpu_start: Pro cpu up.
I (367) cpu_start: Starting app cpu, entry point is 0x40081568
0x40081568: call_start_cpu1 at /Users/taarimi7/Downloads/proj/esp-idf/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (384) cpu_start: Pro cpu start user code
I (385) cpu_start: cpu freq: 160000000 Hz
I (385) cpu_start: Application information:
I (389) cpu_start: Project name:     bleuart5
I (394) cpu_start: App version:      1
I (399) cpu_start: Compile time:     Jan 25 2024 11:34:33
I (405) cpu_start: ELF file SHA256:  1a3d2ec7b5fcad54...
I (411) cpu_start: ESP-IDF:          v5.1.2-dirty
I (416) cpu_start: Min chip rev:     v0.0
I (421) cpu_start: Max chip rev:     v3.99 
I (426) cpu_start: Chip rev:         v1.0
I (431) heap_init: Initializing. RAM available for dynamic allocation:
I (438) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (444) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (450) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (456) heap_init: At 3FFC5508 len 0001AAF8 (106 KiB): DRAM
I (462) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (469) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (475) heap_init: At 4009A450 len 00005BB0 (22 KiB): IRAM
I (483) spi_flash: detected chip: generic
I (486) spi_flash: flash io: dio
W (490) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (504) coexist: coex firmware version: b6d5e8c
I (509) app_start: Starting scheduler on CPU0
I (513) app_start: Starting scheduler on CPU1
mI (513) main_task: Started on CPU0
I (523) main_task: Calling app_main()
I (543) BTDM_INIT: BT controller compile version [ec4ac65]
I (543) BTDM_INIT: Bluetooth MAC: e8:9f:6d:a6:c9:52
I (543) phy_init: phy_version 4780,16b31a7,Sep 22 2023,20:42:16
I (87I (873) NimBLE: BLE server UART_task started

I (873) NimBLE: BLE Host Task Started
I (873) main_task: Returned from app_main()
I (913) NimBLE: Device Address: 
I (913) NimBLE: e8:9f:6d:a6:c9:52
I (913) NimBLE: 

I (913) NimBLE: GAP procedure initiated: advertise; 
I (913) NimBLE: disc_mode=2
I (923) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (933) NimBLE:

dhrishi, 2024-01-29:
@riegaz Is there related to Matter? From the provided details, it doesn't look like that. I would suggest you to close this issue and instead file an issue on https://github.com/espressif/esp-idf

riegaz, 2024-01-29:
Sorry for posting this issue here even though it is ESP-IDF related.

closed 2024-01-29 by riegaz