project-chip/connectedhomeip #31697: [ICD] If chip-tool sends the register-client command, the information is not stored in the ICD storage

mkardous-silabs, 2024-01-25:
#### Description
If the registration with the ICD is done at commissioning, the ICD information is persisted.
If the registration is done after commissioning, the ICD information is not persisted

erjiaqing, 2024-01-26:
> If the registration is done after commissioning, the ICD information is not persisted

I guess this is done by sending the command manually?

I would suggest add a "register" command to icd to do register + persist.

yunhanw-google, 2024-01-29:
@erjiaqing let's go with a new command for registrer-client that can store the icd info to storage.

mkardous-silabs, 2024-04-05:
@erjiaqing @yunhanw-google We would also need an `unregister` command that can clear the persistant storage information.

yunhanw-google, 2024-10-10:
https://github.com/project-chip/connectedhomeip/pull/33671

closed 2024-10-10 by yunhanw-google