project-chip/connectedhomeip #31710: [BUG] nodeId changed during commissioning to 112233

igor90ns, 2024-01-26:
### Reproduction steps

Using latest libs from master v1.2.0.1 for android
I am implementing sdk to my app,  initially it looks that it work fine, but I have found that during commissioning progress node Id that i have set when pair method is called is replaced with 112233 after setting Noc is completed

In my implementation expected behavior is to start from node id 112234 and continue incrementation with every device

in logs I can observe that everything starts ok
ChipDeviceController     D  Pairing device with ID: 112239
onCommissioningStatusUpdate nodeId 112239 stage ReadCommissioningInfo errorCode 0

and then after noc is added node id is changed from 112239 to 112233

MatterDeviceManager      E  onCommissioningStatusUpdate nodeId 112233 stage SendTrustedRootCert errorCode 0

Can you help me to resolve this issue, complete logs are attached

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

181b0cb14ff007ec912f2ba6627e05dfb066c008

### Platform

android

### Platform Version(s)

v1.2.0.1 

### Anything else?

[nodeid.txt](https://github.com/project-chip/connectedhomeip/files/14066285/nodeid.txt)

bzbarsky-apple, 2024-01-27:
@igor90ns What was the node ID in the NOC?

igor90ns, 2024-01-27:
> @igor90ns What was the node ID in the NOC?

I am not currently work, so I don't have access to check it. Noc is received in encoded base64 format.
This noc is decoded to byte array and passed as part of controllerParams in method call onNocGenerated().
From logs that i have attached in this issue noc is:

`MIIB3DCCAYGgAwIBAgIBATAKBggqhkjOPQQDAjAiMSAwHgYKKwYBBAGConwBAwwQMDAwMDAw\r\nMDAwMDAwMDAwMjAgFw0yMDExMDQwMTAyNDJaGA8yMDcwMTAyMzAxMDI0MlowRDEgMB4GCisG\r\nAQQBgqJ8AQUMEDVFNEREQzIzNDhBRDIyOUExIDAeBgorBgEEAYKifAEBDBAwMDAwMDAwMDAw\r\nMDFCNjY5MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEcgTJx5ZQh28H67/CAHjc4ZuQPhXj\r\n+lB40j63y2Zop5Q694YSuzoT4kuwFPvMUIze1KEZ/9/NWGt8HORkktOPdKOBgzCBgDAMBgNV\r\nHRMBAf8EAjAAMA4GA1UdDwEB/wQEAwIHgDAgBgNVHSUBAf8EFjAUBggrBgEFBQcDAgYIKwYB\r\nBQUHAwEwHQYDVR0OBBYEFCet7CoZEXODID4oGNYdNZqXpRjOMB8GA1UdIwQYMBaAFL1wMwF4\r\nAOiidmT9J9cxtoG6kB2NMAoGCCqGSM49BAMCA0kAMEYCIQCgw+JTm6kSamGTlRdfZ20o8bSx\r\nSCMe5GKqoQtXnPi89gIhANHVTVW3ksaZeu1tS6sRysUVMv7NaRIWNNOxdg1mfTP7`

My task is only to implement Android commisioninig part, so  I am not fully  familiar what noc contains. Does noc contains Node id? Can I found it in noc in this format or I need to contact team that worked on task for creating noc.
It would be helpful to know where to search for problem, in android implementation or in received Noc 

Thanks for helping

igor90ns, 2024-01-29:
False alarm, I have checked with team that implemented noc generation, and they hardcoded nodeId.
Now everything works as expected

closed 2024-01-29 by igor90ns

bzbarsky-apple, 2024-02-03:
> Does noc contains Node id?

Sorry for the lag, but yes.  The NOC contains _the_ node ID.  The way the node knows what its node ID is, is by getting a NOC and reading it from there.