project-chip/connectedhomeip #31729: [Feature] How can I detect a Matter Device is using Thread or Wi-Fi

huynguyen-df, 2024-01-29:
### Feature description

# Problem
I'm using. the lib [Matter](https://developer.apple.com/documentation/matter) of Apple.
I have a problem when I want to detect whether the device is using which network connection Wi-Fi or Thread to provide the correct credentials it needs.
Anyone can help me?

### Platform

darwin

bzbarsky-apple, 2024-01-29:
@huynguyen-df You should be using https://developer.apple.com/documentation/mattersupport which should handle that for you, no?

huynguyen-df, 2024-01-29:
Thank you. However, because of some reason, I don't want to apply it to the current project. So I want to find a good other way to detect should use which network connection Wi-Fi or Thread, instead of using MatterSupport to detect that.

bzbarsky-apple, 2024-01-30:
@huynguyen-df Once you have a PASE session to the device, you should be able to get this information from the Network Commissioning cluster, no?

That said, some devices can do both Wi-Fi and Thread; I have no idea what you want to do there.

huynguyen-df, 2024-01-31:
Thank you so much. The Network Commissioning cluster worked well. It correct with my requirement

closed 2024-01-31 by huynguyen-df