project-chip/connectedhomeip #31740: [Build] Android JAR file related questions

shuiYe-704265085, 2024-01-29:
### Build issue(s)

I would like to ask if the source code of the required jar files in the chip-tool Android version is open, such as CHIPController.jar. I am interested in learning about the matter-related onboarding and authentication processes.

### Platform

android

yunhanw-google, 2024-01-31:
yes, they are open, the source code for CHIPController.jar mainly located in 
https://github.com/project-chip/connectedhomeip/tree/master/src/controller/java/src/chip/devicecontroller

https://github.com/project-chip/connectedhomeip/tree/master/src/controller/java

The android chip-tool example code is in 
https://github.com/project-chip/connectedhomeip/tree/master/examples/android/CHIPTool

Feel free to post quetion in github, or reach me(yunhanw-google ) in slack channel, thanks

yunhanw-google, 2024-02-03:
one ways is to generate jar via chip-tool
git clone https://github.com/project-chip/connectedhomeip.git
cd connectedhomeip
git pull
git submodule init
git submodule update
docker run -it -v ~/workspace/connectedhomeip:/connectedhomeip [ghcr.io/project-chip/chip-build-android:33](http://ghcr.io/project-chip/chip-build-android:33)
git config --global --add [safe.directory](http://safe.directory/) "*"
source scripts/bootstrap.sh
./scripts/build/build_examples.py --target android-arm64-chip-tool build

another way to follow https://project-chip.github.io/connectedhomeip-doc/guides/android_building.html