project-chip/connectedhomeip #31786: [Feature] Vendor ID Specific Commissioning of Endpoints

jvmahon, 2024-01-31:
### Feature description

...
Maybe this is already possible (if so, then this is more of a documentation request), but

There are some useful endpoint types like "Generic Switch" and "Mode Select" that are not supported on all Controllers.   In some cases, the controllers produce useless endpoint devices creating cluttered UIs.  As a case in point, If one addes Generic Switch endpoints to a dimmer device to do button tap detection, you have a situation where this is recognized by Apple iOS Home, but if you also commission to Google, Google recognizes the dimmer endpoint, but creates useless dummy devices for the (several) Generic Switches.

This leads to to alternative bad choices - you can leave out the Generic Switch endpoints, in which case you reduce the functionaltiy on some controllers for the sake of avoiding UI messes on others, or you can include the endponts, in which case you may get functionaltiy sometimes, but a UI mess other times.

As a solution, it would be desirable if, during commissioning, one could detect the vendor ID of the controller that the device is being commissioned to, and based on the vendor ID, the node could present a different list of endpoints. So, for example, in the Generic Switch case, if the vendor ID == Apple, present both dimmer endpoints as well as Generic Switch endpoints, but in the case of vendor ID == Google Home, only present the Dimmer endpoints, but not the Generic Switch.

I think this is a matter of manipulating the Root Device's PartsList attribute based on the vendor ID for of the controller to which a device is being commissioned.

Of course, the better solution would be if controllers supported more of these endpoint types, but that seems to be uncertain and far away.

### Platform

all

bzbarsky-apple, 2024-01-31:
You _can_ detect the vendor ID of the commissioner.  It's right there in the AddNOC command.

But this approach breaks completely as soon as a device is commissioned by two different commissioners, no?  You can't have different PartsList on different simultaneous fabrics; that's a spec violation.

jvmahon, 2024-01-31:
> But this approach breaks completely as soon as a device is commissioned by two different commissioners, no? You can't have different PartsList on different simultaneous fabrics; that's a spec violation.

Yes, this is what I wanted to do. Since you are saying that is a spec violation, I guess the issue to be considered is whether this *should* be a spec violation, or if it should be permissible.

At present, it seems that the way Matter controllers are progressing, it is unlikely that  all controllers will support all the endpoint types and features. So, either device makers have to program for the "least capable common set" or have devices that don't behave right on certain fabrics / controllers..

In an ideal world, controller certification would be "tougher" and require controllers to all implement at least a base set of features like Mode Select, OTA, Generic Switch. I doubt that will happen.

So, there is a problem for the technology and manufacturers - either (a) they have to limit capabilities to make the device work on the range of controllers consumers use, or (b) there needs to be some way to customize endpoints based on the  vendor ID, or you end up with devices that create user interface problems  (as is the issue being dealt with here for Google Home and the Generic Switch type).

So, the request is (i) consider allowing different PartsList depending on fabric, and (ii) add functionality to allow this to be programmed in a device (probably more on an "exception" basis -- i.e, always give full PartsList except if XXX).

bzbarsky-apple, 2024-02-01:
So a few things:

1. This isn't the right venue for spec change proposals; those need to happen in the CSA working group.
2. Consider what happens if devices hardcode these sorts of vendor checks and then controllers (which update a lot more often than devices) are changed to handle the situation better... but those devices continue to not work as well as they could with those controllers.
3. In general, it's hard for certification to test what UI looks like because the spec generally can't require anything about UIs...  those are _really_ context-dependent.

Speaking of item 2: I assume you've reported the relevant issues to the controllers that are not doing the "right thing" here?

closed 2024-06-10 by jvmahon