project-chip/connectedhomeip #31788: [BUG] Home APP status and devices cannot synchronize

RedPoppies, 2024-01-31:
### Reproduction steps

Devices that do not use the Matter protocol and are connected to the Home APP via a bridge may encounter issues with synchronization or unresponsiveness when dynamically adding or triggering sub-devices.

### Bug prevalence

indeterminacy，Approximately one week

### GitHub hash of the SDK that was being used

83f4163b1fba35bcf7454806c48c8f84b53517a6

### Platform

darwin

### Platform Version(s)

1.1-branch

RedPoppies, 2024-01-31:
The approximate translation is: "Approximately divided into the following four categories of issues.
1. When triggering the reporting of temperature and humidity on RelativeHumidityMeasurement and TemperatureMeasurement devices, the Home APP may display that the device is unresponsive. This situation may eventually recover after some time, but other types of devices may also experience synchronization issues.

2. There is a small probability of being unable to delete a device when removing the bridge device from the Home APP.

3. When adding a bridge device to the Home APP, using the emberAfSetDynamicEndpoint interface to dynamically add a device may result in the newly added device not synchronizing with the Home APP.

4. There is a probability that the DoorLock added to the Home APP may have issues with issuing unlock commands. A video is attached for reference.

bzbarsky-apple, 2024-01-31:
@RedPoppies This sounds like an issue for Home app, not a core Matter SDK issue.  Please file a Feedback Assistant ticket (or several, for the several different issues observed, since they might need different logs) following the instructions below and mention the ticket IDs here, and I will have those followed up on.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

closed 2025-04-02 by woody-apple