project-chip/connectedhomeip #31791: [BUG] Matter firmware may crash on failed packet buffer allocation

Damian-Nordic, 2024-01-31:
### Reproduction steps

Due to the code added in https://github.com/project-chip/connectedhomeip/pull/28800/files#diff-78ab891d6949886efbe161233c4cce22034eb7497625e647f51637318dda38d2R471, `CommandHandler::AddStatus` crashes if a packet buffer allocation fails.

It's difficult to provide exact reproduction steps but if, say, `onoff on` command is sent to a device while there are multiple ongoing conversations, the device may crash when trying to allocate a buffer to send the command status back to the controller:
```
[00:11:59.027,343] <inf> chip: [ZCL]Endpoint 1 On/off already set to new value
[00:11:59.027,435] <err> chip: [CSL]PacketBuffer: pool EMPTY.
abort()
[00:11:59.027,496] <err> os: r0/a1:  0x00000004  r1/a2:  0x00000000  r2/a3:  0x0000003a
[00:11:59.027,496] <err> os: r3/a4:  0x20005bb8 r12/ip:  0x00000000 r14/lr:  0x0002a74d
...
```

Note sure why the code does `VerifyOrDie` but I guess it's better to just skip a response than crash in such a situation, right?

### Bug prevalence

I received one report of that so far

### GitHub hash of the SDK that was being used

d38a6496c3abeb3daf6429b1f11172cfa0112a1a

### Platform

nrf, core

### Anything else?

Tested on `d38a6496c3abeb3daf6429b1f11172cfa0112a1a` but the code is still there in the current TOT:
```
void CommandHandler::AddStatus(const ConcreteCommandPath & aCommandPath, const Protocols::InteractionModel::Status aStatus,
                               const char * context)
{
    // Return early in case of requests targeted to a group, since they should not add a response.
    VerifyOrReturn(!IsGroupRequest());
    VerifyOrDie(FallibleAddStatus(aCommandPath, aStatus, context) == CHIP_NO_ERROR);
}
```

Damian-Nordic, 2024-01-31:
@andy31415 fyi

bzbarsky-apple, 2024-01-31:
@tcarmelveilleux

Damian-Nordic, 2024-02-21:
@andy31415 could you comment why `VerifyOrDie` has been added and would you be OK with relaxing this check to ignore the `CHIP_ERROR_NO_MEMORY` error? I can propose a fix but wanted to understand the rationale.

andy31415, 2024-03-05:
We can change the code to use `FallibleAddStatus` and just log the error.
However then we would need a large comment explaining why it is ok to receive a command, process it and then fail to send the reply to the other side.

Damian-Nordic, 2024-03-05:
I would argue that it's better that the default be ignoring the error instead of crashing.

`VerifyOrDie` should only be used in scenarios that the device might enter unrecoverable state or hit a condition that it didn't expect. Temporary lack of resources is not an instance of these. While I agree that not sending a response is certainly a bad thing, rebooting in this case is like fighting fire with gasoline :D as then we not only don't send a response but also tear down all sessions and subscriptions and put the device offline until all the controllers re-subscribe.

If a given cluster is able to handle such an error, it can use `FallibleAddStatus` but in a typical scenario, not much can be done when we run out of resources, except for logging an error and maybe increasing some error counter (which we don't have). Note that a packet can also be dropped silently by lower layers due to running out of buffers or other errors at OpenThread or even MAC layers though MRP makes the impact lower.

An alternative to ignoring `CHIP_ERROR_NO_MEMORY` would be allocating a response buffer upon receiving a request so that the command is not even processed if there's no space to send a response back. But I'm not sure if that's actually a better solution.

PR #32463 (2024-03-07): [app] Do not crash if CommandHandler fails to allocate packet
CommandHandler uses VerifyOrDie when adding a status to be sent to the requestor. If the device runs out packet buffers and CommandHandler fails to allocate a packet for the status, the device crashes.

Triggering the crash requires many commands to arrive in the device around the same time, which is rare but possible.

Fixes #31791
closed 2024-03-07 by mergify[bot]