project-chip/connectedhomeip #31793: [1.2] FAILURE in First read Basic Information Node_Label Attr

emesamwong, 2024-01-31:
### Reproduction steps

The expected result in reading the Node_label attr should be an EMPTY value if not set, but I got FAILURE even though I already set a string. In addition, DUT LOG contains an error:
[image]

****
And I could write success and read back the expected string. 
What should I do to pass the TC-BINFO-2.1 step 17 cert test?
[image]

Chip-tool Log:
[Node_Label_issue.txt](https://github.com/project-chip/connectedhomeip/files/14110582/Node_Label_issue.txt)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

esp-matter(2f692ac10a09ed27c223499ecdc458f7730d544d), connectedhomeip (2f692ac10a09ed27c223499ecdc458f7730d544d)

### Platform

esp32

### Platform Version(s)

v5.1.2 (482a8fb2d78e3b58eb21b26da8a5bedf90623213)

### Type

Common Cluster Logic

bzbarsky-apple, 2024-01-31:
@emesamwong What does your ZAP configuration look like for that attribute?  Can you attach (not paste; it's large) your generated `endpoint_config.h` file to this issue?

`err = 15` is `CHIP_ERROR_INVALID_TLV_CHAR_STRING`, which can really only happen if the string contains embedded nulls, as far as I can tell.

emesamwong, 2024-02-01:
hi, @bzbarsky-apple, Thank you for your reply.
This is the endpoint config file in the esp-matter level, I am not sure if the file is correct or not. please let me know if the file is wrong.
[endpoint_config.txt](https://github.com/project-chip/connectedhomeip/files/14119168/endpoint_config.txt)

I set a no-null string for the node label as the issue mentioned, it is not an embedded nulls string ("Benny") but return the same error.

Here is how esp-matter creates and defines the Node Label:

Test code
[image]
Original code
[image]

[image]

[image]

bzbarsky-apple, 2024-02-01:
> This is the endpoint config file in the esp-matter level

Ah, this is not using the ZAP config for this attribute.

In the code you pasted: you are passing `sizeof(config->node_label)` as the string length.  That's too big, and includes the trailing nulls in the string, so the thing that ends up being attempted to be encoded has embedded nulls.  When the API wants a string length, pass the length, not the size of the whole buffer....

closed 2024-02-01 by bzbarsky-apple