project-chip/connectedhomeip #31804: [Enhancement] Adding logging to Matter to aid in debuggability

woody-apple, 2024-01-31:
### Reproduction steps

In looking through Matter logging, we could use a little more detail to make it easier to triage networks with multiple devices, and generally associate logs together.

Collecting together a group of "simple" log enhancements that would help here.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

current

### Platform

core

woody-apple, 2024-01-31:
```
        ChipLogProgress(ExchangeManager,
                        "Retransmitting MessageCounter:" ChipLogFormatMessageCounter " on exchange " ChipLogFormatExchange
                        " Send Cnt %d",
                        messageCounter, ChipLogValueExchange(&entry->ec.Get()), entry->sendCount);
```
Let's add the nodeID and/or more context so we can figure out what this is retransmitting to

woody-apple, 2024-01-31:
```
    ChipLogError(Discovery, "Mdns: Resolve failure (%s)", errorStr);
```
We should log what we're trying to discover here.

woody-apple, 2024-01-31:
```
    ChipLogError(Controller, "Creating NSError from %" CHIP_ERROR_FORMAT, errorCode.Format());
```
Log which MTRDevice (and nodeID) this is for

woody-apple, 2024-01-31:
```
            ChipLogError(ExchangeManager,
                         "Failed to Send CHIP MessageCounter:" ChipLogFormatMessageCounter " on exchange " ChipLogFormatExchange
                         " sendCount: %u max retries: %d",
                         messageCounter, ChipLogValueExchange(&ec.Get()), sendCount, CHIP_CONFIG_RMP_DEFAULT_MAX_RETRANS);
```
Add nodeID to this

woody-apple, 2024-01-31:
```
    ChipLogError(SecureChannel, "CASESession timed out while waiting for a response from the peer. Current state was %u",
                 to_underlying(mState));
```
We should log the nodeID here.

woody-apple, 2024-01-31:
```
    if (err == CHIP_NO_ERROR)
    {
        ChipLogProgress(Controller, "Successfully finished commissioning step '%s'", StageToString(report.stageCompleted));
    }
    else
    {
        ChipLogProgress(Controller, "Error on commissioning step '%s': '%s'", StageToString(report.stageCompleted), err.AsString());
    }

    if (err != CHIP_NO_ERROR)
    {
        completionStatus.failedStage = MakeOptional(report.stageCompleted);
        ChipLogError(Controller, "Failed to perform commissioning step %d", static_cast<int>(report.stageCompleted));
        if (report.Is<AttestationErrorInfo>())
        {
```
This area in AutoCommissioner (and more down there) to allow us to know which device we're talking to (Could be vid/pid, etc)

bzbarsky-apple, 2024-02-01:
> Let's add the nodeID and/or more context so we can figure out what this is retransmitting to

We could, but the message counter can be used to find the original transmission, which has that info...

> Log which MTRDevice (and nodeID) this is for

That information is not currently available there.  We could add another initializer which allows passing it in, and update all the callsites that have that info, of course.

woody-apple, 2024-02-02:
The key here is to have it on a single line, without manual humans reading through logs trying to associate things

carol-apple, 2024-02-02:
Logging NodeID would be great here too:

MTR_LOG_DEFAULT("Read attribute work item [%llu] report duplicate %@", workItemID, readItem);

Along with all other logging in MTRDevice.mm that begins with "Read attribute work item"

PR #32053 (2024-02-14): Adding logging to Matter to aid in debuggability
Resolves #31804

Trying to make matter logs a little easier to debug :)

Made this PR Darwin only, since we need to urgently debug things.
closed 2024-02-14 by woody-apple