project-chip/connectedhomeip #31817: [Documentation] Queries Regarding Commissioning Matter Bridge From ble->thread

mttrRDRSDX, 2024-02-01:
### Documentation issues

Hi ,
I have couple of queries for which am not getting proper Answer from Documentation.
When use the following command to commission 
pairing ble-thread 101 hex:<dataset> 20202021 3840

lets take an example we are trying to commission matter bridge via chiptool and we have border router config and running

so chiptool will read the ble adv from matter bridge and it will start the commissioning process 
now
query 1 :
 does chiptool informs or in any way communicate with border router that device is about to be commissioned inside the thread network?
 if so then what kind of communication is between border router and chiptool

query 2 : 
Thread Network is a base for matter network ? means is matter network made inside the thread network or its different because without border router when we use the onnetwork commissioning how does matter network is behaving or built ?

jinbiaochenstayhungry, 2024-02-01:
HI,  
    for u query ，this is my understanding，it may be wrong，still need more authoritative answer: 
1:  of course ，chip tool need  to communicate with route， it found device in ble av，and commission in pase through ble， the comminucate  with route  with ipv6 packages(to send commission complete command for example).
2: matter is base on ipv6, so it supports all ethernet and wifi dn thread ,only because they all base on ipv6. matter supports thread well ,but cant say that thread is a bese for matter.

mttrRDRSDX, 2024-02-01:
_

> 1: of course ，chip tool need to communicate with route， it found device in ble av，and commission in pase through ble， the comminucate with route with ipv6 packages(to send commission complete command for example).

_

Okay what you are trying to mention is 
bridge found via BLE 
commissioning in PASE via BLE
then chiptool communicates about bridge to router in ipv6 packets format and thats how border router also gets to know about a new device ?

> > 
> 2: matter is base on ipv6, so it supports all ethernet and wifi dn thread ,only because they all base on ipv6. matter supports thread well ,but cant say that thread is a bese for matter.
> 

see my point 
i was reading on internet as well and i found below stuff :

**Every Matter node begins its participation in a Matter network after it is commissioned (**the initial configuration of the device) by a commissioner (say a phone app that configures the device).
Matter Commissioning
During the process of commissioning, the commissioner will provide the node with the Node Operational Certificate (NOC) and the Trusted Root CA Certificate. This is when we say that the bulb has joined the Matter Fabric green (technically Matter Fabrics have a 64-bit identifier, we are just using green for convenience here).

so what exactly is this matter network that is agreed that it is having a collection of matter nodes but what is this network actually is it subnetwork in thread / wifi kind of thing?

mttrRDRSDX, 2024-02-02:
If i summarize the whole concept it will be like follow

During commissioning phase , 
1. chiptool will find matter bridge via BLE Adv (ble-thread)
2. chiptool will cont. commissioning in PASE over BLE 
3. chiptool communicates with OTBR about commissiioned Device and configures the matter bridge to work in Thread network .
4. and in thread network we have our logical matter network formed at higher layer 

is that fine to summarize the stuff

mttrRDRSDX, 2024-02-06:
Anyone can please confirm the understanding as part of last message ? so that we can close the ticket

jinbiaochenstayhungry, 2024-02-11:
step 3 may be a litter different , after step 2, chip-tool has done send thread infos to bridge app, then   bridge app is the one responsed to communicate with Thread roles like thread route to join the thread ,after that ,communication happens after step 4 in thread technology to control device or update states.