project-chip/connectedhomeip #31821: [1.2] [Android] OnboardingPayloadParser cant parser optionalQRCodeInfo

freemanspace, 2024-02-01:
### Reproduction steps

QRCode data as below

**MT:6ZWJ0E.N16OXN77XG3P04-O-1ILZJ29CZK1VNYE1NPTE1KADO1ZTZR19EUE1ZK8N1JC-L10OK523FFU1Q1HN28K4S21E7G2B1FU1O3VE10.9E32NTO3TXWE1ETE528F0T0**

var payload = OnboardingPayloadParser().parseQrCode(str)
payload.optionalQRCodeInfo is empty

but if  use 1.1.x version use 
SetupPayloadParser.parseQrCode() can get optionalQRCodeInfo

please help fix

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

tag:1.2.0.1

### Platform

android

### Type

Platform Issue

yunhanw-google, 2024-02-02:
@freemanspace  could you try this https://github.com/project-chip/connectedhomeip/pull/31251? which add optional qr code info support

yunhanw-google, 2025-02-06:
this should be resolved, feel free to reopen it if it is problematic. thanks

closed 2025-02-06 by yunhanw-google