project-chip/connectedhomeip #31851: [1.2] [Python] Cannot finish activate.sh

angus19, 2024-02-02:
### Reproduction steps

Step-by-step commands as follows.
- `git clone https://github.com/project-chip/connectedhomeip.git`
- `cd connectedhomeip && git checkout 181b0cb`
- `git submodule update --init`
- `source scripts/activate.sh`

Please check attached log for more details about failing of `source scripts/activate.sh`. 
[matter_console.log](https://github.com/project-chip/connectedhomeip/files/14134497/matter_console.log)

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

181b0cb

### Platform

other

### Platform Version(s)

Ubuntu 22.03.4 LTS

### Type

Platform Issue

### Anything else?

Wipe original hard drive before installing a brand new Ubuntu (22.04.3 LTS).
Follow the build guide to set up a build environment and start a build by referring to the link below.
- https://project-chip.github.io/connectedhomeip-doc/guides/BUILDING.html#build-for-the-host-os-linux-or-macos

Launch `source scripts/activate.sh` and an Assertion failed with related exceptions or traceback messages come out...
>   Setting up Python environment.....[\]
ERROR at /home/angus/projects/connectedhomeip/third_party/pigweed/repo/pw_build/facade.gni:187:7: Assertion failed.
      assert(_dep_is_in_link_dependencies,
      ^-----
/home/angus/projects/connectedhomeip/third_party/pigweed/repo/pw_log:impl must be listed in the pw_build_LINK_DEPS build arg when the /home/angus/projects/connectedhomeip/third_party/pigweed/repo/pw_log:pw_log facade is in use. Please update your toolchain configuration.
See /home/angus/projects/connectedhomeip/third_party/pigweed/repo/pw_log/BUILD.gn:47:1: whence it was called.
pw_facade("pw_log") {
^--------------------
See /home/angus/projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/BUILD.gn:50:5: which caused the file to be included.
    "$dir_pw_log:protos.python",
    ^--------------------------
['gn', 'gen', '/home/angus/projects/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/angus/projects/connectedhomeip/third_party/pigweed/repo"']

> Traceback (most recent call last):
  File "/home/angus/projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 402, in install_packages
    subprocess.check_call(
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['gn', 'gen', '/home/angus/projects/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/angus/projects/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

> During handling of the above exception, another exception occurred:

> Traceback (most recent call last):
  File "/home/angus/projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 995, in <module>
    sys.exit(main())
  File "/home/angus/projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 987, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/angus/projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 577, in setup
    result = step(spin)
  File "/home/angus/projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 766, in virtualenv
    if not virtualenv_setup.install(
  File "/home/angus/projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 443, in install
    install_packages(gn_target)
  File "/home/angus/projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 410, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['gn', 'gen', '/home/angus/projects/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/angus/projects/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.
... ...

Is there still anything missing in my Python configuration?
Thanks for comments if any.
-Angus

dt-atmosic, 2024-02-28:
+1

Same hash, same OS, same error log.

dt-atmosic, 2024-03-02:
gn is tripping over system-absolute vs. source-absolute pathnames in the facade.gni assert.  If you hack pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py to use a source-absolute pathname for pw_root (literally "//third_party/pigweed/repo"), bootstrap and activate can finish.

dt-atmosic, 2024-03-02:
@angus19 check to see if there is a difference between "pwd" and "realpath ." in your environment.  I was able to get past this issue by making those identical.

andy31415, 2024-04-11:
Could you try:

```
git submodule update -f --init --recursive
```

for the submodule update and then try the bootstrap again?

bzbarsky-apple, 2024-04-11:
@dt-atmosic Thank you for figuring out what the issue is!  Fix (or at least workaround) coming up.

ksperling-apple, 2024-04-11:
This sounds like something that should be fixed on the Pigweed side, has an issue been opened there yet?

PR #34477 (2024-07-26): [v1.2] Fix bootstrapping
- Cherry picking #32945 to fix #31851
- Bump the pyyaml version to `6.0.1`
- Bump the construct version to `2.10.70`
<details>
  <summary>error due to pyyaml</summary>
  
    × Getting requirements to build wheel did not run successfully.
    │ exit code: 1
    ╰─> [54 lines of output]
        running egg_info
        writing lib/PyYAML.egg-info/PKG-INFO
        writing dependency_links to lib/PyYAML.egg-info/dependency_links.txt
        writing top-level names to lib/PyYAML.egg-info/top_level.txt

</details>

<details>
    <summary> error due to construct </summary>

 [... cut]
closed 2024-04-12 by mergify[bot]