project-chip/connectedhomeip #31854: [Test Failed] write multiple attributes using chip-tool

hyukdong, 2024-02-02:
### Test issue(s)

Hello everyone,
I have trouble writing multiple attributes using chip-tool.
Please let me know the correct chip-tool command.
Thank you in advance for your consideration.
 
Heres are what I've been working on:
### objectives: writing DoorOpen Events (0x004) which is unsigned integer on Door Lock cluster (0x101) and Node Label (0x005) string on Basic Information cluster (0x0028) with a single request
> #### - Trial 1: 
> ```
> ./chip-tool any write-by-id 0x101,0x28 0x004,0x005 10,\'help\' 1 1,0
> ```
> Result on Trial 1: Json Parsing Error
> ```
> [1706851540.578776][81112:81112] CHIP:TOO: Error parsing JSON for attribute-values:
> [1706851540.578828][81112:81112] CHIP:TOO:     Extra non-whitespace after JSON value.
> [1706851540.578848][81112:81112] CHIP:TOO: 10,'help'
> [1706851540.578869][81112:81112] CHIP:TOO: --^
> ```

> #### - Trial 2: 
> ```
> ./chip-tool any write-by-id 0x101,0x28 0x004,0x5 '{"0x0":1,"0x1":"ab12"}' 1 1,0
> ```
> - Result on Trial 2: both Door Lock and Basic Information clusters get the same Json Data
> ```
> [1706854088.064287][71613:71613] CHIP:DMG: WriteRequestMessage =
> [1706854088.064315][71613:71613] CHIP:DMG: {
> [1706854088.064339][71613:71613] CHIP:DMG:      suppressResponse = false,
> [1706854088.064367][71613:71613] CHIP:DMG:      timedRequest = false,
> [1706854088.064393][71613:71613] CHIP:DMG:      AttributeDataIBs =
> [1706854088.064429][71613:71613] CHIP:DMG:      [
> [1706854088.064455][71613:71613] CHIP:DMG:              AttributeDataIB =
> [1706854088.064485][71613:71613] CHIP:DMG:              {
> [1706854088.064513][71613:71613] CHIP:DMG:                      AttributePathIB =
> [1706854088.064615][71613:71613] CHIP:DMG:                      {
> [1706854088.064653][71613:71613] CHIP:DMG:                              Endpoint = 0x1,
> [1706854088.064688][71613:71613] CHIP:DMG:                              Cluster = 0x101,
> [1706854088.064726][71613:71613] CHIP:DMG:                              Attribute = 0x0000_0004,
> [1706854088.064759][71613:71613] CHIP:DMG:                      }
> [1706854088.064794][71613:71613] CHIP:DMG:
> [1706854088.064843][71613:71613] CHIP:DMG:                      Data =
> [1706854088.064874][71613:71613] CHIP:DMG:                      {
> [1706854088.064909][71613:71613] CHIP:DMG:                              0x0 = 1,
> [1706854088.064947][71613:71613] CHIP:DMG:                              0x1 = "ab12" (4 chars),
> [1706854088.064984][71613:71613] CHIP:DMG:                      },
> [1706854088.065013][71613:71613] CHIP:DMG:              },
> [1706854088.065099][71613:71613] CHIP:DMG:
> [1706854088.065128][71613:71613] CHIP:DMG:              AttributeDataIB =
> [1706854088.065158][71613:71613] CHIP:DMG:              {
> [1706854088.065186][71613:71613] CHIP:DMG:                      AttributePathIB =
> [1706854088.065218][71613:71613] CHIP:DMG:                      {
> [1706854088.065250][71613:71613] CHIP:DMG:                              Endpoint = 0x0,
> [1706854088.065285][71613:71613] CHIP:DMG:                              Cluster = 0x28,
> [1706854088.065320][71613:71613] CHIP:DMG:                              Attribute = 0x0000_0005,
> [1706854088.065352][71613:71613] CHIP:DMG:                      }
> [1706854088.065385][71613:71613] CHIP:DMG:
> [1706854088.065416][71613:71613] CHIP:DMG:                      Data =
> [1706854088.065448][71613:71613] CHIP:DMG:                      {
> [1706854088.065481][71613:71613] CHIP:DMG:                              0x0 = 1,
> [1706854088.065519][71613:71613] CHIP:DMG:                              0x1 = "ab12" (4 chars),
> [1706854088.065553][71613:71613] CHIP:DMG:                      },
> [1706854088.065581][71613:71613] CHIP:DMG:              },
> [1706854088.065613][71613:71613] CHIP:DMG:
> [1706854088.065639][71613:71613] CHIP:DMG:      ],
> [1706854088.065676][71613:71613] CHIP:DMG:
[1706854088.065702][71613:71613] CHIP:DMG:      moreChunkedMessages = false,
[1706854088.065729][71613:71613] CHIP:DMG:      InteractionModelRevision = 11
[1706854088.065754][71613:71613] CHIP:DMG: },

```

### Platform

all

bzbarsky-apple, 2024-02-02:
Looks like the behavior was changed without changing the help text...  The correct separator for attribute values is `;`, not `,`.  So this should work:

```
./chip-tool any write-by-id 0x101,0x28 0x004,0x005 '10;"help"' 1 1,0
```
or this:
```
./chip-tool any write-by-id 0x101,0x28 0x004,0x5 '{"0x0":1,"0x1":"ab12"};20' 1 1,0
```

hyukdong, 2024-02-05:
@bzbarsky-apple 
appreciateyou for your prompt response.
Checked out the both of your command guides work.

closed 2024-02-06 by mergify[bot]