project-chip/connectedhomeip #31903: [BUG] Thermostat SystemModeEnum kDry and kSleep returns fail

pankore, 2024-02-05:
### Reproduction steps

Using chip-tool to write system-mode with SystemModeEnum kDry or KSleep  (e.g., ./chip-tool thermostat write system-mode 8 1 1), returns 0x87 (CONSTRAINT_ERROR).

Seems that it is because of https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/thermostat-server/thermostat-server.cpp#L447, where the Enum is larger than kFanOnly, it returns invalidvalue = constrainterror.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

83f8665f0177906a66b927b1a24e8ac038861c86

### Platform

ameba

samryanuofl, 2024-04-04:
I ran into this as well, I assume that's not the intended behavior?

samryanuofl, 2024-04-04:
Created #32859. One thing I don't understand is how other devices are supposed to know whether Fan Only and Dry options are supported on a device, since they aren't tied to a feature bit.

dusikasss, 2024-12-14:
I've created issue  #36846. It proposes finding a method to detect supported operating modes for room air conditioner devices using the Thermostat cluster in Matter.