project-chip/connectedhomeip #31909: [Build] High Memory Usage During Compilation of Chip-tool or Sample App

Survensa, 2024-02-05:
### Build issue(s)

1. source scripts/bootstrap.sh
2. source scripts/activate.sh
For Chip-tool :
3. scripts/examples/gn_build_example.sh examples/chip-tool out/chip-tool 'chip_mdns="platform" chip_inet_config_enable_ipv4=false'
For Sample app :
4. scripts/examples/gn_build_example.sh      examples/all-clusters-app/linux/      examples/all-clusters-app/linux/out/all-clusters-app   chip_inet_config_enable_ipv4=false 
...

### Platform

raspi

### Anything else?

Memory usage is excessively high, even though I only have the `connectedhomeip `directory in my device. I am unable to successfully build, or it takes an extended amount of time to compile the `chip-tool/Sample app` with 4GB RAM. It requires at least a device with 8GB RAM to achieve the build without encountering issues.

[image]

Survensa, 2024-02-05:
Video clip : Tried to rebuild with same command you can see the build gets hanged in Device with 4GB RAM

https://github.com/project-chip/connectedhomeip/assets/95559513/570165b2-ae4c-46c3-b246-272ee3c5b7f0

bzbarsky-apple, 2024-02-05:
chip-tool has a _very_ large translation unit, which includes headers that cover all commands/attributes in all clusters in Matter.

The high memory use during compilation is expected given the current structure of the code.

Survensa, 2024-02-06:
Swapping memory helped in building without any issues. Hence closing this issue

closed 2024-02-06 by Survensa