project-chip/connectedhomeip #31938: [Test Failed]

lboue, 2024-02-05:
### Test issue(s)

# Matter nRF Connect Generic Switch Example Application
## Group 2: Generic Switch

I would try Group 2 Generic Switch for nRF5340 DK board.
Could you tel me how  to switch to https://github.com/project-chip/connectedhomeip/tree/fea16046a88c0a5e8a9e4b789f0463c9c5f334d1/examples/light-switch-app/nrfconnect#generic-switch mode?

By defaut Group 1 is enabled:
```
I: Button has been pressed, keep in this state for at least 500 ms to change light sensitivity of binded lighting devices.
I: Dimming started...
I: Notify Bounded Cluster | endpoint: 1 cluster: 6
I: Notify Bounded Cluster | endpoint: 1 cluster: 8
I: Notify Bounded Cluster | endpoint: 1 cluster: 8
I: Notify Bounded Cluster | endpoint: 1 cluster: 8
I: Notify Bounded Cluster | endpoint: 1 cluster: 8
```

### Platform

nrf connect

Damian-Nordic, 2024-02-06:
If I recall correctly, the generic switch is enabled by default. It's just on endpoint 2 instead of 1. Have you tried running the procedure from https://github.com/project-chip/connectedhomeip/tree/fea16046a88c0a5e8a9e4b789f0463c9c5f334d1/examples/light-switch-app/nrfconnect#testing-the-generic-switch?

If this doesn't work, then it needs fixing.

lboue, 2024-02-06:
> If I recall correctly, the generic switch is enabled by default. It's just on endpoint 2 instead of 1. Have you tried running the procedure from [...]

I thought I will have logs in the console regarding endpoint 2.
I will try running the procedure. I'll keep you informed of the result.