project-chip/connectedhomeip #31941: [Tooling] why is it taking so long for chip-tool interactive to come up in the ci?

cecille, 2024-02-05:
### Reproduction steps / Feature

See https://github.com/project-chip/connectedhomeip/pull/31893#issuecomment-1927084736. yikes.

### Platform

other

### Type

CI tested

bzbarsky-apple, 2024-02-05:
Just to be clear: right now we don't know whether it's "fails to come up withing 10s" or "fails to come up, period".

It's not 100% clear how to tell those cases apart short of somehow tracking the distribution of number of retries it takes in CI...

cecille, 2024-02-06:
Yup, totally fair. @kliao-csa - do your scripts have any insight here?