project-chip/connectedhomeip #31958: [Android] Ble connection closed after 15 sec while network setup is in progress

igor90ns, 2024-02-06:
### Reproduction steps

I am implementing sdk to my app, using latest libs from master v1.2.0.1 for android

I have problem with updating network credentials method from chip controller. Problem is that when I show dialog to user to confirm ssid and to type in password, in 15 seconds ble connection is closed and i cant send **deviceController.updateCommissioningNetworkCredentials(network)**

1. sdk is initilaized with 
                    ControllerParams.newBuilder()
                        .setControllerVendorId(VENDOR_ID)
                        .setSkipCommissioningComplete(true)
                        .setAttemptNetworkScanWiFi(true)
                        .setAttemptNetworkScanThread(true)
                        .setFabricId(fabricId).build()
                )
2. My flow is to call method below with null for network params
**deviceController.pairDevice(gatt, connId, nextNodeId, chipDeviceInfo.setupPinCode, null)**
this works fine, all commissioning  steps including installing Noc are performed successfully

3. I need to determinate if device is wifi or thread. I am doing this in callback 
**onReadCommissioningInfo(vendorId: Int, productId: Int, wifiEndpointId: Int, threadEndpointId: Int)** 
5. and when onScanNetworksSuccess from ScanNetworksListener is called, then I show Dialog to user to confirm currently connected ssid and to provide wifi password
6. then on confirmation I call **deviceController.updateCommissioningNetworkCredentials(network)**

This works correclty only if user finish password input in 15 seconds, if not Bluetooth connection is closed and  I cant send network params to  updateCommissioningNetworkCredentials 

> ack recv timeout, closing ep 0x75cd8dde28
> Clearing BLE pending packets.
> Received UnsubscribeCharacteristic
> setCharacteristicNotification() - uuid: 
> MATTER-3545.onDescriptorWrite: 
> Auto-closing end point's BLE connection.
> Received CloseConnection
> close()
> unregisterApp() - mClientIf=7
> onCloseBleComplete
> Installing profile for com.rtrk.oblosmarthome

understandably when I call updateCommissioningNetworkCredentials I get

> D updateCommissioningNetworkCredentials() called
D  Setting wifi credentials from parameters
D  Setting attestation nonce from parameters
D  Setting CSR nonce from parameters
E  onCommissioningStatusUpdate nodeId 112314 stage NeedsNetworkCreds errorCode 0
D  Successfully finished commissioning step 'NeedsNetworkCreds'
D  Commissioning stage next step: 'NeedsNetworkCreds' -> 'WiFiNetworkSetup'
D  Performing next commissioning step 'WiFiNetworkSetup'
D  ICR moving to [AddingComm]
D  ICR moving to [AddedComma]
D  <<< [E:9315i S:26711 M:259961046] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [BLE] --- Type 0001:08 (IM:InvokeCommandRequest)
E  Failed to send AddOrUpdateWiFiNetwork command: src/transport/raw/BLE.cpp:102: CHIP Error 0x00000003: Incorrect state
E  onCommissioningStatusUpdate nodeId 112314 stage WiFiNetworkSetup errorCode 3
D  Error on commissioning step 'WiFiNetworkSetup': 'src/transport/raw/BLE.cpp:102: CHIP Error 0x00000003: Incorrect state'
E  Failed to perform commissioning step 23
D  Going from commissioning step 'WiFiNetworkSetup' with lastErr = 'src/transport/raw/BLE.cpp:102: CHIP Error 0x00000003: Incorrect state' -> 'Cleanup'
D  Performing next commissioning step 'Cleanup' with completion status = 'src/transport/raw/BLE.cpp:102: CHIP Error 0x00000003: Incorrect state'
E  onCommissioningStatusUpdate nodeId 112314 stage Cleanup errorCode 0
D  Successfully finished commissioning step 'Cleanup'
E  onCommissioningComplete nodeId 112314 errorCode 3
E  onCommissioningComplete failed

Can anyone help me to overcome this issue. I cant pin point origin of this issue and who is closing my ble connection.

Complete log is attached as well
[matter wifi configuration logs.txt](https://github.com/project-chip/connectedhomeip/files/14181565/matter.wifi.configuration.logs.txt)

### Platform

android

### Platform Version(s)

v1.2.0.1 

### Type

Manually tested with SDK, Platform validated

igor90ns, 2024-03-22:
False alarm, closing issue 
Issue is caused by uncertified Tuya smart plug (vendor id 4701, product id 851). After we bought new devices from different manufacturer, everything works as expected .

closed 2024-03-22 by igor90ns

lurendong, 2024-11-27:
Excuse me, are the other devices working? . In the case of 1.4, I have a problem with devices waiting for about 15 seconds after reading device information, i.e. onReadCommissioningInfo waits for about 15 seconds.
[image]