project-chip/connectedhomeip #31988: [Power Source] Mandatory attribute has to be implemented in sample-app and chip-tool

Survensa, 2024-02-07:
### Feature Area

Other

### Test Case

TC-PS-1.1 and 2.1

### Reproduction steps

**All-clusters-app and Chip-tool has to be enabled with below mentioned attribute as it is mandatory:**

- BatChargePermissions

**Specification Reference:**

[image]

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

1a6bd3b28743a2b2494157190061eeabaf4eb9c3

### Platform

raspi

cecille, 2024-02-09:
Please see spec issue - https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/8158. There is no way this attribute is correctly marked in the spec. None the less, it needs to be fixed before this cluster is launched.

Survensa, 2024-03-20:
Battery management things has been removed from spec in PowerSourceCluster

closed 2024-03-20 by Survensa