project-chip/connectedhomeip #31989: [1.2] Chip-tool Android Application crashed when tried to implement device to device binding

Dhirav-Shah, 2024-02-07:
### Reproduction steps

I perfomed device to device in chip-tool android application.
-  For that first I used AccessControlCluster and after that I used AccessControlClusterAccessControlEntryStruct and passed the values  privilege : 5 , authmode: 2, subjects:[112233] , targets: null , fabricIndex: 1 and after that passed the privilege : 3 , authmode: 2, subjects: <LIGHT SWITCH NODE ID> , targets: null , fabricIndex: 1 for AccessControlClusterAccessControlEntryStruct 

- After that I tried to use writeAclAttribute and passed the values defaultClusterCallback and array of AccessControlClusterAccessControlEntryStruct .

- But at writeAcl app got crashed everytime.

- Please assist me in the same way. I would be grateful for your assistance.

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

Matter

### Platform

android

### Type

Core SDK Crash

yunhanw-google, 2024-02-13:
Could you upload the detailed log and crash dump if possible? thanks

Dhirav-Shah, 2024-02-14:
@yunhanw-google 
     java_vm_ext.cc:591] JNI DETECTED ERROR IN APPLICATION: can't call long java.lang.Long.longValue() on instance of java.lang.Integer
                                                                                                    java_vm_ext.cc:591]     in call to CallLongMethodV
                                                                                                    java_vm_ext.cc:591]     from void chip.devicecontroller.ChipClusters$AccessControlCluster.writeAclAttribute(long, chip.devicecontroller.ChipClusters$DefaultClusterCallback, java.util.ArrayList, java.lang.Integer)
2024-02-14 14:57:41.172 26584-26584 com.eic.matter          com.eic.matter                       A  runtime.cc:691] Runtime aborting...
                                                                                                    runtime.cc:691] Dumping all threads without mutator lock held
                                                                                                    runtime.cc:691] All threads:
                                                                                                    runtime.cc:691] DALVIK THREADS (36):
                                                                                                    runtime.cc:691] "main" prio=10 tid=1 Runnable
                                                                                                    runtime.cc:691]   | group="" sCount=0 ucsCount=0 flags=0 obj=0x720df208 self=0xb4000072b82a47b0
                                                                                                    runtime.cc:691]   | sysTid=26584 nice=-10 cgrp=top-app sched=0/0 handle=0x73fef94500
                                                                                                    runtime.cc:691]   | state=R schedstat=( 5840022869 206233653 2527 ) utm=544 stm=39 core=5 HZ=100
                                                                                                    runtime.cc:691]   | stack=0x7fc19eb000-0x7fc19ed000 stackSize=8188KB
                                                                                                    runtime.cc:691]   | held mutexes= "abort lock" "mutator lock"(shared held)
                                                                                                    runtime.cc:691]   native: #00 pc 004bca80  /apex/com.android.art/lib64/libart.so (art::DumpNativeStack+108) (BuildId: 735f12f804f88d62a2cb437261076ff7)
                                                                                                    runtime.cc:691]   native: #01 pc 004b0dd4  /apex/com.android.art/lib64/libart.so (art::Thread::DumpStack const+388) (BuildId: 735f12f804f88d62a2cb437261076ff7)
                                                                                                    runtime.cc:691]   native: #02 pc 004b0714  /apex/com.android.art/lib64/libart.so (art::DumpCheckpoint::Run+164) (BuildId: 735f12f804f88d62a2cb437261076ff7)
                                                                                                    runtime.cc:691]   native: #03 pc 003ada50  /apex/com.android.art/lib64/libart.so (art::ThreadList::RunCheckpoint+452) (BuildId: 735f12f804f88d62a2cb437261076ff7)
                                                                                                    runtime.cc:691]   native: #04 pc 004c99bc  /apex/com.android.art/lib64/libart.so (art::ThreadList::Dump+232) (BuildId: 735f12f804f88d62a2cb437261076ff7)
                                                                                                    runtime.cc:691]   native: #05 pc 00771fe4  /apex/com.android.art/lib64/libart.so (art::Runtime::Abort+2156) (BuildId: 735f12f804f88d62a2cb437261076ff7)
                                                                                                    runtime.cc:691]   native: #06 pc 000357d0  /apex/com.google.mainline.primary.libs@341177000/lib64/libbase.so/75d3253827fcfd7a8d7b02ad45991611ec4ca424c0278e13e8acfad4d14e597a3ecff6c0caa2b785c73838528ee6e9c2b313240ff895f50ee39b1d7bc10f390a/libbase.so (android::base::SetAborter::$_0::__invoke+80) (BuildId: 6f67f69ff36b970d0b831cfdab3b578d)
                                                                                                    runtime.cc:691]   native: #07 pc 00034d58  /apex/com.google.mainline.primary.libs@341177000/lib64/libbase.so

yunhanw-google, 2024-02-15:
@Dhirav-Shah It may crash here, 
                            for (jint i_3 = 0; i_3 < element_0_subjectsItem_1Size; ++i_3)
                            {
                                jobject element_3;
                                chip::JniReferences::GetInstance().GetListItem(element_0_subjectsItem_1, i_3, element_3);
                                listHolder_3->mList[i_3] = static_cast<std::remove_reference_t<decltype(listHolder_3->mList[i_3])>>(
                                    chip::JniReferences::GetInstance().LongToPrimitive(element_3));
                            }
What UI are you using, are you using wilcard UI or cluster Interaction UI, it seems subject 112233 from app is set as Integer instead of long, how do you fill subjects:[112233]? Could you try to set without subjects? I think it may not be crash.
thanks

Mitun90, 2024-02-16:
hi @yunhanw-google,

I've successfully developed device-to-device binding in an iOS application. 

I now want to be able to recognize my newly installed devices, such as "light-switch-app" or "lighting-app" in chip tool iOS application. Is there a way for us to obtain device identifying information or cluster information from the device payload?

yunhanw-google, 2024-02-17:
for supported cluster, you can get it from descriptor cluster.
for device identify info, you can get it from identify cluster.

@Mitun90 has the above android issue resolved in your side in android? What UI are you using, are you using wilcard UI or cluster Interaction UI, it seems subject 112233 from app is set as Integer instead of long, how do you fill subjects:[112233]? Could you try to set without subjects? 

thanks

yunhanw-google, 2024-02-17:
@Dhirav-Shah  are you using cluster Interaction Tool UI? 
https://github.com/project-chip/connectedhomeip/blob/c13b324a3b9dcd3b278015c03c33ceab5fbd1a30/examples/android/CHIPTool/app/src/main/java/com/google/chip/chiptool/clusterclient/clusterinteraction/ClusterDetailFragment.kt#L200C15-L200C31
It seems we cannot write a array using android chip-tool.

i guess you are using the API for this binding cluster, then you have to make sure 112233 is not long when you declare its type.

Mitun90, 2024-02-19:
> [ue resolved in your side in android?](https://github.com/project-chip/connectedhomeip/issues/31989#issuecomment-1949536169)

@yunhanw-google yes. android issue now resolved and successfully bind "light-switch-app" and "lighting-app" in android mobile application.

-> we used cluster interaction UI
-> For subjects:[112233], we tried without subjects and it works successfully.

Thank you.

Dhirav-Shah, 2024-02-19:
@yunhanw-google  thanks for your replies.

yunhanw-google, 2024-02-22:
May I know the exact scenario and identify what property? Thanks

yunhanw-google, 2024-02-22:
I would close this one, feel free to recreate another issue with the scenario you wnana know about identify cluster, thanks

closed 2024-02-22 by yunhanw-google