project-chip/connectedhomeip #31990: [CERT-TEST-FAILURE] Extended Announcement fails in linux

Saravana-kr22, 2024-02-07:
### Feature Area

Other

### Test Case

TC-DD-2.1

### Reproduction steps

Step 1: Advertise the Dut: ` ./chip-all-clusters-app`
Step-2: Wait for 15 mins

### Bug prevalence

Whenever i do this 

### GitHub hash of the SDK that was being used

1a6bd3b28743a2b2494157190061eeabaf4eb9c3

### Platform

raspi

### Anything else?

The Extended Announcement is still not implemented in the linux platform, After waiting for 15 mins after the advertisement has been stopped.

> [1707315228.744046][47431:47431] CHIP:SVR: Closing pairing window
[1707315228.744168][47431:47431] CHIP:IN: SecureSession[0xaaaae1d6a700]: Released - Type:1 LSID:19179
[1707315228.744234][47431:47431] CHIP:DIS: Updating services using commissioning mode 0
[1707315228.765469][47431:47431] CHIP:DIS: CHIP minimal mDNS started advertising.
[1707315228.770948][47431:47431] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on otbr0:  src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1707315228.770989][47431:47431] CHIP:DIS: mDNS broadcast full failed in 1 separate send attempts.
[1707315228.771008][47431:47431] CHIP:DIS: Failed to advertise records: src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1707315228.791268][47431:47431] CHIP:DIS: Warning: Attempt to mDNS broadcast failed on otbr0:  src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1707315228.791311][47431:47431] CHIP:DIS: mDNS broadcast full failed in 1 separate send attempts.
[1707315228.791332][47431:47431] CHIP:DIS: Failed to advertise records: src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1707315228.807656][47431:47432] CHIP:DL: UnregisterAdvertisement complete
[1707315228.808298][47431:47431] CHIP:DL: HandlePlatformSpecificBLEEvent 16394
[1707315228.808335][47431:47431] CHIP:DL: CHIPoBLE advertising stopped

Note: Also tried with experimental features enabled in BlueZ

> ubuntu@ubuntu:/connectedhomeip$ sudo systemctl edit bluetooth.service
Editing "/etc/systemd/system/bluetooth.service.d/override.conf" canceled: temporary file is empty.
ubuntu@ubuntu:/connectedhomeip$ sudo systemctl restart bluetooth.service

Still facing the same issue.

arkq, 2024-02-07:
Extended advertising is disabled by default. In order to enable it you have to change these settings:

https://github.com/project-chip/connectedhomeip/blob/f658ea2f316cc0d4f1bdaae96ddd6be3d747a881/src/include/platform/CHIPDeviceConfig.h#L629
https://github.com/project-chip/connectedhomeip/blob/f658ea2f316cc0d4f1bdaae96ddd6be3d747a881/src/include/platform/CHIPDeviceConfig.h#L718

E.g.:
```c++
#define CHIP_DEVICE_CONFIG_BLE_EXT_ADVERTISING 1
#define CHIP_DEVICE_CONFIG_DISCOVERY_TIMEOUT_SECS (20 * 60)  // 5 minutes of extended advertising
```

Saravana-kr22, 2024-02-08:
@arkq After updating this config file the extended announcement is enabled. But  test-plan expect that

```
DUT must send ADVs at a rate equal to 1200ms.

The following Advertising PDU payload fields shall comply with section 5.4.2.5.6. "Advertising Data". Verify that:

12-bit "Discriminator" field in Matter service data is populated

16-bit "Vendor ID" field in Matter service data is set to 0

16-bit "Product ID" field in Matter service data is set to 0

"Extended Data" bit is set to 1 if Extended Data exists otherwise set to 0 (Bit[0] of last byte)

"Extended Announcement" bit is set to 1 in Matter service data (Bit[1] of last byte)
```

But in all-cluster-app the ADV's rate is `1284ms` which is out of range and the "Extended Announcement"  bit needs to 1 but i got the 2 from the 16-bit data.

[image]

So this make the test-step to fail , Please correct me if my understanding is wrong