project-chip/connectedhomeip #32008: [1.3] T key in TXT record set incorrectly

raul-marquez-csa, 2024-02-07:
### Reproduction steps

The T key is set incorrectly per the new spec
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/secure_channel/Discovery.adoc#4-common-txt-keyvalue-pairs

[image]

The restriction on bit 0 is present in the recirculation ballot and hence should be landing in 1.3

### Bug prevalence

n/a

### GitHub hash of the SDK that was being used

master

### Platform

python

### Type

Spec Compliance Issue

bzbarsky-apple, 2024-02-08:
The spec is new and the SDK has not been updated to it yet.  The reason bit 0 (the one that's set here) is reserved in the spec is precisely because the SDK has been setting it and so it means nothing.

@pidarped

raul-marquez-csa, 2024-02-13:
Addressed in:

https://github.com/project-chip/connectedhomeip/pull/32083

closed 2024-02-13 by raul-marquez-csa