project-chip/connectedhomeip #32014: [TC-IDM-1.4] - The Python script should be updated in accordance with the latest changes in the Test Plan.

Rajashreekalmane, 2024-02-08:
### Feature Area

Other

### Test Case

[TC-IDM-1.4] 

### Reproduction steps

The Python script should be updated in accordance with the latest changes in the Test Plan.

Please refer to the Test Plan PR that has been updated for the relevant test case : https://github.com/CHIP-Specifications/chip-test-plans/pull/3937

**Python Script** 
https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_IDM_1_4.py

**Test-Plan Reference :**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-1-4-batched-commands-invoke-response-action-from-dut-to-th-dut_server

### Bug prevalence

Whenevre I do

### GitHub hash of the SDK that was being used

7be9d886ffc50abcbc9fd96abad1cf5c2cab933a

### Platform

raspi

tehampson, 2024-02-09:
PR is waiting for https://github.com/project-chip/connectedhomeip/pull/31979/ to be approved by non-google person to be able to be merged

tehampson, 2024-02-13:
This is fixed since  #31979  is merged

closed 2024-02-13 by tehampson