project-chip/connectedhomeip #32037: [1.2] All-cluster-app pairing with chip tool

salvio96, 2024-02-09:
### Reproduction steps

We have a version of all-cluster-app , which is built from an sdk that contains the matter toolchian built from the build.gn present here https://github.com/project-chip/connectedhomeip/blob/v1.2.0.1 /BUILD.gn.
We took the following source and header files of the all-cluster-app example and the other useful example folders as necessary, such as examle/tracing example/platform/linux, example/persistend and example/provider, we brought with us then the folder generated by zap tools and the callback-stub.cpp and PluginApplicationCallbacks.h files too .

we have the following error:
when we try to pair, the commission command fails because we have the following error:
Failed to read BasicCommissioningInfo: src/app/ClusterStateCache.cpp:270: CHIP Error 0x000000CA: Interaction Model Error
[1706866858.595449][125120:125122] CHIP:CTL: Failed to read RegulatoryConfig: src/app/ClusterStateCache.cpp:270: CHIP Error 0x000000CA: Interaction Model Error
[1706866858.595452][125120:125122] CHIP:CTL: Failed to read LocationCapability: src/app/ClusterStateCache.cpp:270: CHIP Error 0x000000CA: Interaction Model Error

We noticed that the attributes that cause us errors are declared as extrernal

I will attach also the txt file log  , with the run of the chip-tool and all-cluster-app.

[all-cluster.app.log.md](https://github.com/project-chip/connectedhomeip/files/14222397/all-cluster.app.log.md)
[chip-tool-log.txt](https://github.com/project-chip/connectedhomeip/files/14222398/chip-tool-log.txt)

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/commit/0b9dfec117994815c56fe673c48be9352656dcb2

### Platform

other

### Platform Version(s)

1.2

### Type

Core SDK Memory Issue

bzbarsky-apple, 2024-02-09:
In your `GeneralCommissioningAttrAccess::ReadIfSupported`, what error is being returned from the call into the configuration manager for these attributes?  Chances are, your configuration manager is not set up correctly.

salvio96, 2024-02-12:
Apparently, this function is never called.

bzbarsky-apple, 2024-02-13:
Is `GeneralCommissioningAttrAccess::Read` being called?  Is `MatterGeneralCommissioningPluginServerInitCallback` being called?

salvio96, 2024-02-13:
Both are not called.

bzbarsky-apple, 2024-02-14:
OK, then `MatterGeneralCommissioningPluginServerInitCallback` not being called is your problem; you are not properly initializing the General Commissioning cluster.

What does your `PluginApplicationCallbacks.h` look like, and specifically, is `MatterGeneralCommissioningPluginServerInitCallback` included in the definition of `MATTER_PLUGINS_INIT`?

Also, your file listing above seems to list `PluginApplicationCallbacks.h` twice, so it's possible the "wrong" one is being used?

salvio96, 2024-02-15:
Hi , this is  PluginApplicationCallback.h : 

#pragma once
void MatterAccessControlPluginServerInitCallback();
void MatterActionsPluginServerInitCallback();
void MatterActivatedCarbonFilterMonitoringPluginServerInitCallback();
void MatterAdministratorCommissioningPluginServerInitCallback();
void MatterAirQualityPluginServerInitCallback();
void MatterBallastConfigurationPluginServerInitCallback();
void MatterBarrierControlPluginServerInitCallback();
void MatterBasicInformationPluginServerInitCallback();
void MatterBinaryInputBasicPluginServerInitCallback();
void MatterBindingPluginServerInitCallback();
void MatterBooleanStatePluginServerInitCallback();
void MatterCarbonDioxideConcentrationMeasurementPluginServerInitCallback();
void MatterCarbonMonoxideConcentrationMeasurementPluginServerInitCallback();
void MatterColorControlPluginServerInitCallback();
void MatterDescriptorPluginServerInitCallback();
void MatterDiagnosticLogsPluginServerInitCallback();
void MatterDishwasherAlarmPluginServerInitCallback();
void MatterDishwasherModePluginServerInitCallback();
void MatterElectricalMeasurementPluginServerInitCallback();
void MatterEthernetNetworkDiagnosticsPluginServerInitCallback();
void MatterFanControlPluginServerInitCallback();
void MatterFaultInjectionPluginServerInitCallback();
void MatterFixedLabelPluginServerInitCallback();
void MatterFlowMeasurementPluginServerInitCallback();
void MatterFormaldehydeConcentrationMeasurementPluginServerInitCallback();
void MatterGeneralCommissioningPluginServerInitCallback();
void MatterGeneralDiagnosticsPluginServerInitCallback();
void MatterGroupKeyManagementPluginServerInitCallback();
void MatterGroupsPluginServerInitCallback();
void MatterHepaFilterMonitoringPluginServerInitCallback();
void MatterIcdManagementPluginServerInitCallback();
void MatterIdentifyPluginServerInitCallback();
void MatterIlluminanceMeasurementPluginServerInitCallback();
void MatterLaundryWasherControlsPluginServerInitCallback();
void MatterLaundryWasherModePluginServerInitCallback();
void MatterLevelControlPluginServerInitCallback();
void MatterLocalizationConfigurationPluginServerInitCallback();
void MatterLowPowerPluginServerInitCallback();
void MatterModeSelectPluginServerInitCallback();
void MatterNetworkCommissioningPluginServerInitCallback();
void MatterNitrogenDioxideConcentrationMeasurementPluginServerInitCallback();
void MatterOccupancySensingPluginServerInitCallback();
void MatterOnOffPluginServerInitCallback();
void MatterOnOffSwitchConfigurationPluginServerInitCallback();
void MatterOperationalCredentialsPluginServerInitCallback();
void MatterOperationalStatePluginServerInitCallback();
void MatterOtaSoftwareUpdateRequestorPluginServerInitCallback();
void MatterOzoneConcentrationMeasurementPluginServerInitCallback();
void MatterPm10ConcentrationMeasurementPluginServerInitCallback();
void MatterPm1ConcentrationMeasurementPluginServerInitCallback();
void MatterPm25ConcentrationMeasurementPluginServerInitCallback();
void MatterPowerSourcePluginServerInitCallback();
void MatterPowerSourceConfigurationPluginServerInitCallback();
void MatterPressureMeasurementPluginServerInitCallback();
void MatterPumpConfigurationAndControlPluginServerInitCallback();
void MatterRadonConcentrationMeasurementPluginServerInitCallback();
void MatterRefrigeratorAlarmPluginServerInitCallback();
void MatterRefrigeratorAndTemperatureControlledCabinetModePluginServerInitCallback();
void MatterRelativeHumidityMeasurementPluginServerInitCallback();
void MatterRvcCleanModePluginServerInitCallback();
void MatterRvcOperationalStatePluginServerInitCallback();
void MatterRvcRunModePluginServerInitCallback();
void MatterScenesPluginServerInitCallback();
void MatterSmokeCoAlarmPluginServerInitCallback();
void MatterSoftwareDiagnosticsPluginServerInitCallback();
void MatterSwitchPluginServerInitCallback();
void MatterTemperatureControlPluginServerInitCallback();
[... 70 lines cut ...]
    MatterRadonConcentrationMeasurementPluginServerInitCallback(); \
    MatterRefrigeratorAlarmPluginServerInitCallback(); \
    MatterRefrigeratorAndTemperatureControlledCabinetModePluginServerInitCallback(); \
    MatterRelativeHumidityMeasurementPluginServerInitCallback(); \
    MatterRvcCleanModePluginServerInitCallback(); \
    MatterRvcOperationalStatePluginServerInitCallback(); \
    MatterRvcRunModePluginServerInitCallback(); \
    MatterScenesPluginServerInitCallback(); \
    MatterSmokeCoAlarmPluginServerInitCallback(); \
    MatterSoftwareDiagnosticsPluginServerInitCallback(); \
    MatterSwitchPluginServerInitCallback(); \
    MatterTemperatureControlPluginServerInitCallback(); \
    MatterTemperatureMeasurementPluginServerInitCallback(); \
    MatterThermostatPluginServerInitCallback(); \
    MatterThermostatUserInterfaceConfigurationPluginServerInitCallback(); \
    MatterThreadNetworkDiagnosticsPluginServerInitCallback(); \
    MatterTimeFormatLocalizationPluginServerInitCallback(); \
    MatterTimeSynchronizationPluginServerInitCallback(); \
    MatterTotalVolatileOrganicCompoundsConcentrationMeasurementPluginServerInitCallback(); \
    MatterUnitLocalizationPluginServerInitCallback(); \
    MatterUnitTestingPluginServerInitCallback(); \
    MatterUserLabelPluginServerInitCallback(); \
    MatterWakeOnLanPluginServerInitCallback(); \
    MatterWiFiNetworkDiagnosticsPluginServerInitCallback(); \
    MatterWindowCoveringPluginServerInitCallback();

bzbarsky-apple, 2024-02-16:
OK, in that case is `emberAfInit` being called?

salvio96, 2024-02-19:
Yes this is the output:
the first line is a debug line added in util.cpp to see the called of the function

 ########EmberInit  
[1708331657.606841][11690:11690] CHIP:DMG: Failed to read stored attribute (0, 0x0000_0028, 0x0000_0005: matter/app/src/Linux/KeyValueStoreManagerImpl.cpp:53: CHIP Error 0x000000A0: Value not found in the persisted storage
[1708331657.606853][11690:11690] CHIP:DMG: Failed to read stored attribute (0, 0x0000_0028, 0x0000_0010: matter/app/src/Linux/KeyValueStoreManagerImpl.cpp:53: CHIP Error 0x000000A0: Value not found in the persisted storage
[1708331657.606862][11690:11690] CHIP:DMG: Failed to read stored attribute (0, 0x0000_002B, 0x0000_0000: matter/app/src/Linux/KeyValueStoreManagerImpl.cpp:53: CHIP Error 0x000000A0: Value not found in the persisted storage
[1708331657.606871][11690:11690] CHIP:DMG: Failed to read stored attribute (0, 0x0000_002C, 0x0000_0000: matter/app/src/Linux/KeyValueStoreManagerImpl.cpp:53: CHIP Error 0x000000A0: Value not found in the persisted storage
[1708331657.606880][11690:11690] CHIP:DMG: Failed to read stored attribute (0, 0x0000_002C, 0x0000_0001: matter/app/src/Linux/KeyValueStoreManagerImpl.cpp:53: CHIP Error 0x000000A0: Value not found in the persisted storage
[1708331657.606889][11690:11690] CHIP:DMG: Failed to read stored attribute (0, 0x0000_002D, 0x0000_0000: matter/app/src/Linux/KeyValueStoreManagerImpl.cpp:53: CHIP Error 0x000000A0: Value not found in the persisted storage

salvio96, 2024-02-19:
Just an update: I've missed MATTER_PLUGINS_INIT flag definition used in EmberAfInit(), Now I added it including #include <app/PluginApplicationCallbacks.h>, so now void MatterGeneralCommissioningPluginServerInitCallback() function is called but it crashes with this error:

[1708343305.098015][96225:96225] CHIP:DMG: AccessControlCluster: initializing
[1708343305.098017][96225:96225] CHIP:ZCL: Initiating Admin Commissioning cluster.
[1708343305.098020][96225:96225] CHIP:-: ==== DEBUG ====: MatterGeneralCommissioningPluginServerInitCallback() works
[1708343305.098022][96225:96225] CHIP:DMG: Endpoint 0, Cluster 0x0000_0030 update version to 130c54bc
[1708343305.098029][96225:96225] CHIP:ZCL: GeneralDiagnostics: OnDeviceReboot
[1708343305.098031][96225:96225] CHIP:DMG: Endpoint 0, Cluster 0x0000_0033 update version to e219381d
[1708343305.098046][96225:96225] CHIP:EVL: LogEvent event number: 0x0000000000020000 priority: 2, endpoint id:  0x0 cluster id: 0x0000_0033 event id: 0x3 Epoch timestamp: 0x0000018DC132238A
[1708343305.098059][96225:96225] CHIP:ZCL: ScenesServer::Instance().Init() error: /home/user/matterclient/platform/matter/app/src/all-cluster-app/clusters/scenes-server/SceneTableImpl.cpp:582: CHIP Error 0x00000003: Incorrect state

Thread 1 "matter" received signal SIGSEGV, Segmentation fault.
0x00000000003fbbb9 in chip::TimeSyncDataProvider::Load (this=<optimized out>, key=<optimized out>, buffer=...)
    at /home/user/matterclient/platform/matter/app/src/all-cluster-app/clusters/time-synchronization-server/TimeSyncDataProvider.cpp:224
224	    ReturnErrorOnFailure(mPersistentStorage->SyncGetKeyValue(key, buffer.data(), size));

salvio96, 2024-02-21:
I've done some step forward: now I tried to pair app with chip-tool it stops at Commissioning step 14

[1708510958.104938][72731:72731] CHIP:ZCL: OpCreds: Got FailSafeTimerExpired
[1708510958.104942][72731:72731] CHIP:ZCL: OpCreds: Proceeding to FailSafeCleanup on fail-safe expiry!
 
Thread 1 "matter" received signal SIGSEGV, Segmentation fault.
chip::FabricInfo::Reset (this=0x4e37d8 <chip::Server::sServer+4136>)
    at ../../src/credentials/FabricTable.h:205
205	            chip::Platform::Delete(mOperationalKey);
(gdb) backtrace
#0  chip::FabricInfo::Reset() (this=0x4e37d8 <chip::Server::sServer+4136>)
    at ../../src/credentials/FabricTable.h:205
#1  0x000000000048a8ea in chip::FabricTable::RevertPendingOpCertsExceptRoot()
    (this=0x4e2e58 <chip::Server::sServer+1704>) at ../../src/credentials/FabricTable.cpp:2034
#2  0x0000000000488531 in chip::FabricTable::RevertPendingFabricData()
    (this=0x4e2e58 <chip::Server::sServer+1704>) at ../../src/credentials/FabricTable.cpp:2013
#3  0x000000000036ad6e in (anonymous namespace)::FailSafeCleanup(chip::DeviceLayer::ChipDeviceEvent const*)
    (event=0x7fffffffdf00)
    at /home/monteiro/platform/matter/app/src/all-cluster-app/clusters/operational-credentials-server/operational-credentials-server.cpp:298
#4  (anonymous namespace)::OnPlatformEventHandler(chip::DeviceLayer::ChipDeviceEvent const*, long)
    (event=0x7fffffffdf00, arg=<optimized out>)
    at /home/monteiro/platform/matter/app/src/all-cluster-app/clusters/operational-credentials-server/operational-credentials-server.cpp:319
#5  0x00000000003a996d in chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToApplication(chip::DeviceLayer::ChipDeviceEvent const*)
    (this=0x4dc0d0 <chip::DeviceLayer::PlatformManagerImpl::sInstance+16>, event=0x7fffffffdf00)
    at /home/monteiro/platform/matter/app/src/include/platform/internal/GenericPlatformManagerImpl.ipp:336
#6  chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*)
    (this=this@entry=0x4dc0d0 <chip::DeviceLayer::PlatformManagerImpl::sInstance+16>, event=event@entry=0x7fffffffdf00)
    at /home/monteiro/platform/matter/app/src/include/platform/internal/GenericPlatformManagerImpl.ipp:301
#7  0x00000000003a9d06 in chip::DeviceLayer::PlatformManager::DispatchEvent(chip::DeviceLayer::ChipDeviceEvent const*) (this=0x4dc0c0 <chip::DeviceLayer::PlatformManagerImpl::sInstance>, event=0x7fffffffdf00)
    at /home/monteiro/.docker-build-scripts/conan_cache/.conan/data/connectedhomeip/1.2.0.1/conan_toolchain_catalog/build/package/123a906d5c14ecd54b85b99247603ecc9f69ffcb/include/platform/PlatformManager.h:503
#8  chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::ProcessDeviceEvents() (this=0x4dc0d0 <chip::DeviceLayer::PlatformManagerImpl::sInstance+16>)
    at /home/monteiro/platform/matter/app/src/include/platform/internal/GenericPlatformManagerImpl_POSIX.ipp:182
#9  chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() (this=<optimized out>)
    at /home/monteiro/platform/matter/app/src/include/platform/internal/GenericPlatformManagerImpl_POSIX.ipp:224
#10 0x0000000000397bc8 in chip::DeviceLayer::PlatformManager::RunEventLoop() (this=<optimized out>)
    at /home/monteiro/.docker-build-scripts/conan_cache/.conan/data/connectedhomeip/1.2.0.1/conan_toolchain_catalog/build/package/123a906d5c14ecd54b85b99247603ecc9f69ffcb/include/platform/PlatformManager.h:403
#11 ChipLinuxAppMainLoop(AppMainLoopImplementation*) (impl=impl@entry=0x0)
    at /home/monteiro/platform/matter/app/src/platform_linux/AppMain.cpp:605
#12 0x000000000038f20e in main(int, char**) (argc=<optimized out>, argv=<optimized out>)
    at /home/monteiro/platform/matter/app/src/all-cluster-app/main.cpp:36

bzbarsky-apple, 2024-02-22:
> Now I added it including #include <app/PluginApplicationCallbacks.h>

Um...  Is that not included in `src/app/util/util.cpp` already?  I am not sure what you changed here and why you had to.

It's hard to say anything intelligent about your situation if there are random code modifications that I don't know about....

salvio96, 2024-02-22:
Hi no , there aren't random ,code , for the include yes it was in util but it was commented , we leave the comment . Now the problem is that wrote before in the fabriTable index:
 AccessControl: checking f=0 a=p s=0xFFFFFFFB00000000 t= c=0x0000_003E e=0 p=a
[1708591585.373035][3806:3806] CHIP:DMG: AccessControl: implicit admin (PASE)
[1708591585.373041][3806:3806] CHIP:DMG: Received command for Endpoint=0 Cluster=0x0000_003E Command=0x0000_0004
[1708591585.373050][3806:3806] CHIP:ZCL: OpCreds: Received a CSRRequest command
[1708591585.373056][3806:3806] CHIP:ZCL: OpCreds: Finding fabric with fabricIndex 0x0
[1708591585.373095][3806:3806] CHIP:EM: <<< [E:14879r S:37109 M:234243256 (Ack:53757721)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::56ff:e11a:6794:b7c8%wlo1]:37420] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1708591585.373128][3806:3806] CHIP:EM: Flushed pending ack for MessageCounter:53757721 on exchange 14879r
[1708591585.373139][3806:3806] CHIP:ZCL: OpCreds: AllocatePendingOperationalKey returned src/credentials/FabricTable.cpp:1569: CHIP Error 0x00000071: Invalid Fabric Index
[1708591585.373147][3806:3806] CHIP:DMG: Endpoint=0 Cluster=0x0000_003E Command=0x0000_0004 status 0x01 (FAILURE) (no additional context)
[1708591585.373158][3806:3806] CHIP:DMG: Command handler moving to [ Preparing]
[1708591585.373166][3806:3806] CHIP:DMG: Command handler moving to [AddingComm]
[1708591585.373174][3806:3806] CHIP:DMG: Command handler moving to [AddedComma]
[1708591585.373181][3806:3806] CHIP:ZCL: OpCreds: Failed CSRRequest request with IM error 0x01 (err = src/credentials/FabricTable.cpp:1569: CHIP Error 0x00000071: Invalid Fabric Index)
[1708591585.373196][3806:3806] CHIP:DMG: Decreasing reference count for CommandHandler, remaining 0
[1708591585.373226][3806:3806] CHIP:EM: <<< [E:14879r S:37109 M:234243257 (Ack:53757721)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::56ff:e11a:6794:b7c8%wlo1]:37420] --- Type 0001:09 (IM:InvokeCommandResponse)
[1708591585.373263][3806:3806] CHIP:DMG: Command handler moving to [CommandSen]
[1708591585.373271][3806:3806] CHIP:DMG: Command handler moving to [AwaitingDe]
[1708591585.373299][3806:3806] CHIP:EM: >>> [E:14878r S:37109 M:53757722 (Ack:234243255)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1708591585.373305][3806:3806] CHIP:EM: Found matching exchange: 14878r, Delegate: (nil)
[1708591585.373313][3806:3806] CHIP:EM: Rxd Ack; Removing MessageCounter:234243255 from Retrans Table on exchange 14878r
[1708591585.373692][3806:3806] CHIP:EM: >>> [E:14880r S:37109 M:53757723] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
[1708591585.373702][3806:3806] CHIP:EM: Handling via exchange: 14880r, Delegate: 0x6989c8
[1708591585.373716][3806:3806] CHIP:DMG: InvokeRequestMessage =
[1708591585.373721][3806:3806] CHIP:DMG: {
[1708591585.373724][3806:3806] CHIP:DMG: 	suppressResponse = false, 
[1708591585.373728][3806:3806] CHIP:DMG: 	timedRequest = false, 
[1708591585.373733][3806:3806] CHIP:DMG: 	InvokeRequests =
[1708591585.373740][3806:3806] CHIP:DMG: 	[
[1708591585.373746][3806:3806] CHIP:DMG: 		CommandDataIB =
[1708591585.373753][3806:3806] CHIP:DMG: 		{
[1708591585.373759][3806:3806] CHIP:DMG: 			CommandPathIB =
[1708591585.373766][3806:3806] CHIP:DMG: 			{
[1708591585.373773][3806:3806] CHIP:DMG: 				EndpointId = 0x0,
[1708591585.373780][3806:3806] CHIP:DMG: 				ClusterId = 0x30,
[1708591585.373787][3806:3806] CHIP:DMG: 				CommandId = 0x0,
[1708591585.373794][3806:3806] CHIP:DMG: 			},
[1708591585.373803][3806:3806] CHIP:DMG: 			
[1708591585.373811][3806:3806] CHIP:DMG: 			CommandFields = 
[1708591585.373819][3806:3806] CHIP:DMG: 			{
[1708591585.373828][3806:3806] CHIP:DMG: 				0x0 = 0, 
[1708591585.373835][3806:3806] CHIP:DMG: 				0x1 = 0, 
[1708591585.373842][3806:3806] CHIP:DMG: 			},
[1708591585.373848][3806:3806] CHIP:DMG: 		},
[1708591585.373857][3806:3806] CHIP:DMG: 		
[1708591585.373862][3806:3806] CHIP:DMG: 	],
[1708591585.373871][3806:3806] CHIP:DMG: 	
[1708591585.373875][3806:3806] CHIP:DMG: 	InteractionModelRevision = 11
[1708591585.373879][3806:3806] CHIP:DMG: },
[1708591585.373893][3806:3806] CHIP:DMG: AccessControl: checking f=0 a=p s=0xFFFFFFFB00000000 t= c=0x0000_0030 e=0 p=a
[1708591585.373898][3806:3806] CHIP:DMG: AccessControl: implicit admin (PASE)
[1708591585.373903][3806:3806] CHIP:DMG: Received command for Endpoint=0 Cluster=0x0000_0030 Command=0x0000_0000
[1708591585.373909][3806:3806] CHIP:FS: GeneralCommissioning: Received ArmFailSafe (0s)
[1708591585.373914][3806:3806] CHIP:FS: Fail-safe timer expired
[1708591585.373920][3806:3806] CHIP:DMG: Command handler moving to [ Preparing]
[1708591585.373925][3806:3806] CHIP:DMG: Command handler moving to [AddingComm]
[1708591585.373930][3806:3806] CHIP:DMG: Command handler moving to [AddedComma]
[1708591585.373939][3806:3806] CHIP:DMG: Decreasing reference count for CommandHandler, remaining 0
[1708591585.373963][3806:3806] CHIP:EM: <<< [E:14880r S:37109 M:234243258 (Ack:53757723)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::56ff:e11a:6794:b7c8%wlo1]:37420] --- Type 0001:09 (IM:InvokeCommandResponse)
[1708591585.373991][3806:3806] CHIP:DMG: Command handler moving to [CommandSen]
[1708591585.373996][3806:3806] CHIP:DMG: Command handler moving to [AwaitingDe]
[1708591585.374001][3806:3806] CHIP:DL: HandlePlatformSpecificBLEEvent 32782
[... cut ...]
[1708591585.375961][3806:3806] CHIP:DIS: Responding with _CM._sub._matterc._udp.local
[1708591585.375970][3806:3806] CHIP:DIS: Responding with 35BF258DEAE26540._matterc._udp.local
[1708591585.375974][3806:3806] CHIP:DIS: CHIP minimal mDNS configured as 'Commissionable node device'; instance name: 35BF258DEAE26540.
[1708591585.377194][3806:3806] CHIP:DIS: mDNS service published: _matterc._udp
[1708591585.377202][3806:3806] CHIP:ZCL: OpCreds: Got FailSafeTimerExpired
[1708591585.377207][3806:3806] CHIP:ZCL: OpCreds: Proceeding to FailSafeCleanup on fail-safe expiry!

bzbarsky-apple, 2024-02-22:
>  [1708591585.373909][3806:3806] CHIP:FS: GeneralCommissioning: Received ArmFailSafe (0s)

Well, that's the commissioner expiring the fail-safe.  I strongly suggest attaching, not pasting, logs from both the commissioner and the commissionee here.  That might be enough information to figure out why it did that.