project-chip/connectedhomeip #32050: EPM: RMS* and NeutralCurrent attribute value ranges cannot be negative

littlebriars, 2024-02-09:
By definition these parameters cannot be negative, so the constraint ranges should be changed from (-2^62 to 2^62) to (0 to 2^62).

bzbarsky-apple, 2024-02-10:
@littlebriars This is an issue for the spec, not the SDK, right?  Is this filed in the wrong place?

littlebriars, 2024-02-10:
> By definition these parameters cannot be negative, so the constraint ranges should be changed from (-2^62 to 2^62) to (0 to 2^62).

Whoops: yes.

I have now raised a spec issue: https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/8868

For the test spec, I've already included the fix in https://github.com/CHIP-Specifications/chip-test-plans/pull/3949/, which has just been merged.

For the SDK I noticed this in review of https://github.com/project-chip/connectedhomeip/pull/31616, and suggested it be fixed there, but @jamesharrow is not surprisingly reluctant to include it until there is a spec change too, so suggest we keep this issue open until the SDK is fixed (either in #31616 or if it misses that PR an new PR).