project-chip/connectedhomeip #32097: [CERT-TEST-FAILURE] TC-DLOG-2.1 - BDX initiation issues

sumaky, 2024-02-13:
### Feature Area

Other

### Test Case

TC-DLOG-2.1

### Reproduction steps

command given in all-cluster-app:
**./chip-all-clusters-app --end_user_support_log ~/tmp/end_user_support_log.txt --network_diagnostics_log ~/tmp/nw_log.txt --crash_log ~/tmp/crash_log.txt**

command given in CHIP tool:
Note: crash_log.txt is of size 1028 bytes

**./chip-tool diagnosticlogs retrieve-logs-request 2 1 1 0 --TransferFileDesignator "~/tmp/crash_log.txt"**

**Response Obtained:**
  **RetrieveLogsResponse: {
   status: 4
   logContent: 
 }**

Expected Response:
BDX initiation
Status SUCCESS

Actual Behavior:
BDX initiation does not happen
Seeing Denied(4) status instead of status SUCCESS

Is there any other setting required to get this working?

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

dbf3dadad29c1aa3b28e5be05d29e28c3e561618

### Platform

other

### Anything else?

THLOG:
[THLOG.odt](https://github.com/project-chip/connectedhomeip/files/14267915/THLOG.odt)
DUT LOG:
[DUT-LOG.odt](https://github.com/project-chip/connectedhomeip/files/14267929/DUT-LOG.odt)
Sample Crash Log
[crash_log.txt](https://github.com/project-chip/connectedhomeip/files/14267934/crash_log.txt)

vivien-apple, 2024-02-13:
In the TH logs there is the lines:
```
[1707831465.257536][10823:10826] CHIP:TOO: The file '/tmp/~/tmp/crash_log.txt' for dumping the logs does not exist.
[1707831465.257543][10823:10826] CHIP:-: ../../examples/chip-tool/commands/common/BDXDiagnosticLogsServerDelegate.cpp:75: CHIP Error 0x00000003: Incorrect state at src/protocols/bdx/BdxTransferServer.cpp:160
```

In short `chip-tool` does not want to overwrite files. You need to manually removes the file `/tmp/~/tmp/crash_log.txt` yourself first. Can you double check that it helps ?

Also you can have more logs by launching both `chip-tool` and the `all-clusters-app` with `--trace_decode 1`

cjandhyala, 2024-02-14:
@sumaky you need to run the test in chip-tool interactive mode

sumaky, 2024-02-14:
@vivien-apple  deleting the file helps and now we are able to get the SUCCESS Status code. Do we need to consider this as an issue ?

sumaky, 2024-02-14:
Also found couple of issues 
1. if we send TransferFileDesignator with n chars the length is considered as  n-2 char in the log as shown below
actual length is 28 but log shows 26 actual length is 28 but log shows 26
1707909961.024038][38061:38061] CHIP:atm: SendInit
[1707909961.024044][38061:38061] CHIP:atm:   Proposed Transfer Control: 0x10
[1707909961.024050][38061:38061] CHIP:atm:   Range Control: 0x0
[1707909961.024055][38061:38061] CHIP:atm:   Proposed Max Block Size: 1024
[1707909961.024059][38061:38061] CHIP:atm:   Start Offset: 0x0000000000000000
[1707909961.024063][38061:38061] CHIP:atm:   Proposed Max Length: 0x0000000000000000
**[1707909961.024070][38061:38061] CHIP:atm:   File Designator Length: 26
[1707909961.024076][38061:38061] CHIP:atm:   File Designator: "Length_1234567891234.txt"**

sumaky, 2024-02-14:
./chip-tool diagnosticlogs retrieve-logs-request 1 1 1 0 --TransferFileDesignator "nw_log.txt"
the above command returns Exhausted(1) but expected status is NoLogs(2)

sumaky, 2024-02-14:
There is no issue as the file name is inclusive of the quotes
The log file is empty the status code can be exhausted and if the log file is missing the status can be NoLogs(0)

closed 2024-02-14 by sumaky