project-chip/connectedhomeip #32108: [Platform] [Feature] Have MTRStructs conform to NSSecureCoding

moe-mehrabi, 2024-02-13:
### Reproduction steps

It'd be great to have these classes conform to NSSecuredCoding so the client can encode/decode the path directly for storage and xpc communications.

### Platform

darwin

### Platform Version(s)

N/A

### Type

Manually tested with SDK

bzbarsky-apple, 2024-02-14:
@moe-mehrabi Which exact structs are we talking about?  Issue talks about "the path", but all the path interfaces already implement NSSecureCoding.

moe-mehrabi, 2024-02-14:
Update the description of the issue.

bzbarsky-apple, 2024-02-15:
Per Slack discussion: this is pretty dangerous to do if someone then starts using NSSecureCoding to persist these things, because they can change (grow new members, and possibly other changes).

For the specific "send across XPC" use case @moe-mehrabi has, using response-value dictionaries is a better fit.