project-chip/connectedhomeip #32165: [Feature] Battery level alert function

mmt-10, 2024-02-16:
### Feature description

I try to make the thread device for matter lighting, lighting-switch, door-lock, window covering and etc.
There power source are battery AA or AAA.
When battery power is lower than threshold, I would like to send alert to controller.

### Platform

core

jvmahon, 2024-02-16:
I think this can be done by adding the "Power Source Cluster" as a endpoint to your device. See Section 11.7 of the Matter 1.2 Core Specification, Attributes 0x000C, 0x000D, 0x000F with the FeatureMap Setting (Section 11.7.4) set to 0b00000010 = "2" for Battery.

bzbarsky-apple, 2024-02-16:
Yep, adding a power source device to one of your endpoints is the way to go here.

closed 2024-02-16 by bzbarsky-apple