project-chip/connectedhomeip #32173: Spring2024 TE2 - cannot commission 'onnetwork' within docker, but can outside

jamesharrow, 2024-02-16:
### Describe the bug

Trying to run TC_ACE_2.1 apparently you can't run this in the TH UI, and we have to run it (according to slack by running in docker container).

chip-repl/python etc cannot discover the device over avahi, but I can discover it and commission it with chip-tool outside the container (and inside - see updated note below). We had similar issues in TE1 (with a different DUT) but that issue was eventually closed out because they thought it was related to chip-tool.

### Steps to reproduce the behavior

1. Launch container:

ubuntu@ubuntu:~/certification-tool/backend/test_collections/sdk_tests/sdk_checkout/python_testing/scripts/sdk$ **docker run -v /home/ubuntu/certification-tool/backend/test_collections/sdk_tests/sdk_checkout/python_testing:/root/temp_python_testing -v /var/paa-root-certs:/paa_roots -v /var/run/dbus:/var/run/dbus -v $(pwd):/launch_dir --privileged --network host -it connectedhomeip/chip-cert-bins:0ff709f25f6a7af68314c1b0a147a76c30c234f6**

**2. check you can see the DUT in a commissioning mode:**
root@ubuntu:~# avahi-browse -r _matterc._udp

+  wlan0 IPv6 1376D5C8D19257C1                              _matterc._udp        local
+  wlan0 IPv6 024757C781D958B1                              _matterc._udp        local
+  wlan0 IPv6 17DEEB0E3A2F9AEE                              _matterc._udp        local
=  wlan0 IPv6 1376D5C8D19257C1                              _matterc._udp        local
   hostname = [7284AEDA1ADB0672.local]
   address = [2a02:c7c:747a:1a00:be6e:76ff:fe6e:f142]
   port = [5540]
   txt = ["PI=" "PH=33" "CM=1" "D=1171" "T=1" "SAI=2000" "SII=800" "VP=4240+4098"]
=  wlan0 IPv6 024757C781D958B1                              _matterc._udp        local
   hostname = [7284AEDA1ADB0672.local]
   address = [2a02:c7c:747a:1a00:be6e:76ff:fe6e:f142]
   port = [5540]
   txt = ["PI=" "PH=33" "CM=1" "D=1171" "T=1" "SAI=2000" "SII=800" "VP=4240+4098"]
=  wlan0 IPv6 17DEEB0E3A2F9AEE                              _matterc._udp        local
   hostname = [7284AEDA1ADB0672.local]
   address = [2a02:c7c:747a:1a00:be6e:76ff:fe6e:f142]
   port = [5540]
   txt = ["PI=" "PH=33" "CM=1" "D=1171" "T=1" "SAI=2000" "SII=800" "VP=4240+4098"]

NOTE IT HAS IPv6

**3. Try to run the command to run TC_ACE_2.1 per slack comments:**

root@ubuntu:~# cd ~/temp_python_testing/scripts/sdk/

root@ubuntu:~/temp_python_testing/scripts/sdk# python3 TC_AccessChecker.py --commissioning-method on-network --discriminator 1171 --passcode 64889390 --storage-path admin_storage.json --paa-trust-store-path /paa_roots --tests test_TC_ACE_2_1

[1707513696.130090][11:11] CHIP:CTL: Setting attestation nonce to random value

[1707513696.132616][11:11] CHIP:CTL: Setting CSR nonce to random value

[1707513696.134322][11:11] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs

[1707513696.134763][11:11] CHIP:DL: writing settings to file (/tmp/chip_kvs-CmpzRz)

[1707513696.135031][11:11] CHIP:DL: renamed tmp file to file (/tmp/chip_kvs)

[1707513696.135345][11:11] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini

[1707513696.135692][11:11] CHIP:DL: writing settings to file (/tmp/chip_factory.ini-uuA7V9)

[1707513696.135971][11:11] CHIP:DL: renamed tmp file to file (/tmp/chip_factory.ini)

[1707513696.136147][11:11] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini

[1707513696.136421][11:11] CHIP:DL: writing settings to file (/tmp/chip_config.ini-1sOTPy)

[1707513696.136703][11:11] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)

[1707513696.137069][11:11] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini

[1707513696.137335][11:11] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-2dJ3QA)

[1707513696.137537][11:11] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)

[... 236 lines cut ...]
INFO:chip.TS:Reverted Last Known Good Time to previous value

INFO:chip.DL:writing settings to file (/tmp/chip_counters.ini-jqRO9u)

INFO:chip.DL:renamed tmp file to file (/tmp/chip_counters.ini)

INFO:chip.DL:NVS set: chip-counters/total-operational-hours = 0 (0x0)

ERROR:chip.DL:Inet Layer shutdown

ERROR:chip.DL:BLE shutdown

ERROR:chip.DL:System Layer shutdown

ERROR:root:Final result: FAIL !

root@ubuntu:~/temp_python_testing/scripts/sdk# 

**4. exit from the container**
run chip-tool or chip-repl and the DUT commissions ok.

### Environment

Raspberry PI 8GB
Image updated using commandline instructions from previous TE

jamesharrow, 2024-02-29:
This issue seems to be still present in SVE test harness - and will block testing.
Can we escalate the priority of this - to me it seems fine on the SDK (I can run these commands outside of the test harness) but running under the test harness seems to still have issues.

rquidute, 2024-02-29:
Hi @jamesharrow could you please provide the Logs, also could you run `./scripts/th/th-doctor.sh --complete` and provide the output?

jamesharrow, 2024-03-03:
I have changed my network setup (I reflashed the router with OpenWRT) and since then I haven't had this issue and was able to commission the DUT ok with the test harness.

So likely we can mark this as closed now.

closed 2024-03-06 by cecille