project-chip/connectedhomeip #32190: [ICD] Add ActiveModeDuration in ICD Client Storage,

yunhanw-google, 2024-02-16:
the user case is that when controller, for example, chip-tool/android chip-tool queue IM operations, once check-in message comes,  controller flush IM operations one by one, then stay active based upon ActiveModeDuration, after ActiveModeDuration expires or timeout for subscription happens assume you run subscription with LIT, controller would do automatically queue

bzbarsky-apple, 2024-02-17:
> then stay active based upon ActiveModeDuration, after ActiveModeDuration expires or timeout for subscription happens assume you run subscription with LIT, controller would do automatically queue

I don't understand what this is trying to say.

turon, 2024-03-04:
WG: how one client with one ecosystem can send back-to-back to a single ICDs.  Needs to know how long the ICD will stay up, but calculated remaining time requires `ActiveModeThreshold` which is not being stored on client side.

yunhanw-google, 2024-04-30:
closed it since we don't need this any more

closed 2024-04-30 by yunhanw-google