project-chip/connectedhomeip #32208: [CERT-TEST-FAILURE] [TC-IDM-1.3] - Test failed

Rajashreekalmane, 2024-02-19:
### Feature Area

Other

### Test Case

TC-IDM-1.3

### Reproduction steps

- 

1. Command to build chip-repl in connectedhomeIP:
`scripts/run_in_build_env.sh './scripts/build_python.sh --install_virtual_env out/venv -d true'`

2. After successfully building chip-repl, enter the chip-repl.
3. Launch the all-clusters-app by running the following command:
`./chip-all-clusters-app`

4. In the chip-repl, send the command below to initiate the commissioning process:
`devCtrl.CommissionOnNetwork(0x12344321, 20202021)`

5. After successful commissioning, attempt to perform step 3 as mentioned in the test plan:

> TH Client send FailAtFault command to FaultInjection cluster to TH device (Server). FailAtFault arguments are: Type: 3 (ChipFault)
> Id: 12, NumberCallsToSkip: 3, NumCallsToFail:1, TakeMutex: False

**By sending below mentioned command, got the response as UnsupportedCommand (0x81)**

`**await devCtrl.SendCommand(0x12344321, 0, chip.clusters.FaultInjection.Commands.FailAtFault(3,12,3,1,0))**`

> File ~/Feb_cntr/connectedhomeip/out/venv/lib/python3.10/site-packages/chip/ChipDeviceCtrl.py:950, in ChipDeviceControllerBase.SendCommand(self, nodeid, endpoint, payload, responseType, timedRequestTimeoutMs, interactionTimeoutMs, busyWaitMs, suppressResponse)
>     942 device = self.GetConnectedDeviceSync(nodeid, timeoutMs=interactionTimeoutMs)
>     943 ClusterCommand.SendCommand(
>     944     future, eventLoop, responseType, device.deviceProxy, ClusterCommand.CommandPath(
>     945         EndpointId=endpoint,
>    (...)
>     948     ), payload, timedRequestTimeoutMs=timedRequestTimeoutMs,
>     949     interactionTimeoutMs=interactionTimeoutMs, busyWaitMs=busyWaitMs, suppressResponse=suppressResponse).raise_on_error()
> --> 950 return await future
> InteractionModelError: InteractionModelError: UnsupportedCommand (0x81)
> 

- 

The YMAL Script missed to provide the sample command 

[image]

**YMAL Script Reference :** https://github.com/project-chip/connectedhomeip/blob/master/src/app/tests/suites/certification/Test_TC_IDM_1_3.yaml

**Test-plan Reference :** https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-1-3-batched-commands-invoke-request-action-from-dut-to-th-dut_client

### Bug prevalence

whenever I do 

### GitHub hash of the SDK that was being used

c4688b86cdb9345e4c4f12f8086b08ee237de555

### Platform

raspi

tehampson, 2024-02-19:
As stated in the [testplan here](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#required-devices-2):

>Test harness as device which is the recipient of the Invoke Request Message - Server. This device needs nlfaultinject enabled at compile time.

You are not using the nlfaultinject version of the all-clusters-app. This is the problem. This was added in TH image [over here](https://github.com/project-chip/connectedhomeip/pull/32179) so if you are using the latest TH image you need to use `chip-all-clusters-app-nlfaultinject` as the TH server.

If you are compiling this yourself you need to add `-nlfaultinject` to the build command. For example `./scripts/build/build_examples.py --target linux-x64-all-clusters-no-ble-clang-nlfaultinject build`. Make sure that you use the `out/` directory that contains the name `-nlfaultinject`

tehampson, 2024-02-19:
@Rajashreekalmane Please take a look at the above comment and retry with the proper instructions

Rajashreekalmane, 2024-02-19:
Thank you for the clear instruction @tehampson, Script is working as expected