project-chip/connectedhomeip #32218: Does chip-tool provide a cluster or method to "getCurrentFabricIndex"?

whqxbs, 2024-02-20:
### Test issue(s)

In my research on the Matter for Android, I noticed that the fabric index is used to determine if a fabric "**isCurrent**" to the app.

I didn't find a command to query the index of the current fabric(**getCurrentFabricIndex**) when using chip-tool (e.g. command `chip-tool operationalcredentials read fabrics ${NODE_ID} 0 --fabric-filtered false`) the result doesn't mention the current fabric, so I have to make my own judgment.

Does chip-tool provide a cluster or method to "getCurrentFabricIndex"?

Looking forward to your reply! :)

bzbarsky-apple, 2024-02-21:
@whqxbs Does
```
chip-tool operationalcredentials read current-fabric-index ${NOD_ID} 0
```
not do what you want?

bzbarsky-apple, 2024-02-21:
And in particular, are you trying to get the fabric index of the fabric on the _server_ or on the _client_?

whqxbs, 2024-02-21:
> @whqxbs Does
> 
> ```
> chip-tool operationalcredentials read current-fabric-index ${NODE_ID} 0
> ```
> 
> not do what you want?

As shown in the figure I have queried the fabric list of a certain matter device with the chiptool command `chip-tool operationalcredentials read current-fabric-index ${NODE_ID} 0`, but I don't have a way to tell directly which is the fabric of the **chiptool** that is **currently in use**, so the result doesn't say directly what the `FabricIndex` of the **current chiptool** is.
[image]

whqxbs, 2024-02-21:
> And in particular, are you trying to get the fabric index of the fabric on the _server_ or on the _client_?

I'm guessing server? Let me describe a scenario where I've paired a Matter device (called B) with a chiptool (called A) and I'm looking at B's fabric info on A.

bzbarsky-apple, 2024-02-22:
Then
 ```
chip-tool operationalcredentials read current-fabric-index ${NODE_ID} 0
```
should do what you want.

whqxbs, 2024-02-22:
It works! Thank you!

closed 2024-02-22 by bzbarsky-apple