project-chip/connectedhomeip #32219: [TC-CNET-4.4] Python script is not matching with test plan

Survensa, 2024-02-20:
### Feature Area

Other

### Test Case

TC-CNET-4.4

### Reproduction steps

v

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

2c8d0bb9d1c7ac9570317c9d7561e86e3bad4b24

### Platform

raspi

cecille, 2024-02-20:
Please see https://github.com/CHIP-Specifications/chip-test-plans/pull/3991, looks like the test plan PR did not get merged yet. Original test plan was not actually testable at all. If you turn off the access point, you can't send a command so I'm not sure how that was actually being tested.

Survensa, 2024-02-21:
Test plan fixed and now python script is matching with the test plan.

closed 2024-02-21 by Survensa