project-chip/connectedhomeip #32260: [Documentation] NetworkCommissioning cluster scnNetwork SSID min length different from 1.2 Specs

Apollon77, 2024-02-22:
### Documentation issues

The 1.2 Specifications defines 1..32 as length for the SSID parameter when not null or absent.

The chip XML https://github.com/project-chip/connectedhomeip/blob/8fbe0a71deb96ade73282558a583386b80dbcd5a/src/app/zap-templates/zcl/data-model/chip/network-commissioning-cluster.xml#L147C1-L147C97 only defined the maximum length which implies minimum length = 0

Also Amazon is sending such a ScanNetworks request with an existing but 0 length SSID

### Platform

other

bzbarsky-apple, 2024-03-05:
From the specification, section "7.18.1. Nullable":

> Composite data types that have a length (i.e. octet string and list), and derived types that have those as the base type, SHALL NOT differentiate semantically between the null value and the empty (zero length) value. In particular a zero-length value SHALL be allowed for nullable values of these types no matter what other length constraints are imposed on the value, and SHALL have the same semantics as the null value.

This field is nullable, so an empty string (1) must be allowed by the server no matter what constraints say and (2) must mean the same thing as null.  The code in the network commssioning cluster explicitly handles this 0-length case and even references the above-quoted spec bits (though the spec section numbering changed, so it's referencing the wrong spec section 7.17.1).

closed 2024-03-05 by bzbarsky-apple

Apollon77, 2024-03-06:
Ok, thank you for this clarification, we will adjust the code generation on our side