project-chip/connectedhomeip #32273: [CERT-TEST-FAILURE] chip-bridge-app in Matter 1.2 TH fails wildcard subscriptions

Emill, 2024-02-22:
### Feature Area

Other

### Test Case

TC-BR-4

### Reproduction steps

Our DUT (commissioner/controller/client) uses a wildcard subscription to maintain information and state of the server devices.
Our client relies on the information obtained through subscription report data to pass the various test steps of TC-BR-4.
However, the chip-bridge-app, as present on the Matter 1.2 TH (matter-th-fall2023.shrunk.img) seems to have a bug and hang when a wildcard subscription is requested.

On the Raspberry Pi test harness flashed with the above image, the issue can be reproduced by the following steps:
1. Run `rm -rf /tmp/chip_*` to factory reset.
2. Run `./chip-bridge-app` in a separate terminal and keep it running.
3. Run `./chip-tool pairing code 1 34970112332` to pair.
4. Run `./chip-tool any subscribe-by-id 0xFFFFFFFF 0xFFFFFFFF 1 10 1 0xFFFF` to try a wildcard subscription.

A few data attributes are printed but the initial report data never finishes and after six seconds, chip-tool prints "Timeout" and exits with non-zero status code 1.

The chip-bridge-app prints the following in the log as the last things:
```
...
[1708627673.684663][2734:2734] CHIP:DMG: <RE:Run> Cluster 2f, Attribute fffd is dirty
[1708627673.684680][2734:2734] CHIP:DMG: Reading attribute: Cluster=0x0000_002F Endpoint=6 AttributeId=0x0000_FFFD (expanded=1)
[1708627673.684697][2734:2734] CHIP:DMG: AccessControl: checking f=1 a=c s=0x000000000001B669 t= c=0x0000_002F e=6 p=v
[1708627673.684715][2734:2734] CHIP:DMG: AccessControl: allowed
[1708627673.684734][2734:2734] CHIP:DMG: <RE:Run> Cluster 2f, Attribute fff8 is dirty
[1708627673.684750][2734:2734] CHIP:DMG: Reading attribute: Cluster=0x0000_002F Endpoint=6 AttributeId=0x0000_FFF8 (expanded=1)
[1708627673.684767][2734:2734] CHIP:DMG: AccessControl: checking f=1 a=c s=0x000000000001B669 t= c=0x0000_002F e=6 p=v
[1708627673.684784][2734:2734] CHIP:DMG: AccessControl: allowed
[1708627673.684812][2734:2734] CHIP:DMG: Error retrieving data from clusterId: 0x0000_002F, err = src/lib/core/TLVWriter.cpp:633: CHIP Error 0x00000024: Invalid TLV tag
[1708627673.685195][2734:2734] CHIP:DL: writing settings to file (/tmp/chip_kvs-9JrU36)
[1708627673.685979][2734:2734] CHIP:DL: renamed tmp file to file (/tmp/chip_kvs)
[1708627673.687021][2734:2734] CHIP:DMG: IM RH moving to [AwaitingDestruction]
[1708627673.687216][2734:2734] CHIP:EM: <<< [E:40266r S:56746 M:110210123 (Ack:193327468)] (S) Msg TX to 1:000000000001B669 [8C1C] [UDP:[fe80::e65f:1ff:fe87:d5bf%eth0]:44307] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1708627673.687354][2734:2734] CHIP:EM: Flushed pending ack for MessageCounter:193327468 on exchange 40266r
```

After this, the subscription "hangs" and no further messages are printed relating to this subscription. The bridge app however continues to process following, new, requests as usual.

Chip-tool output:
```
...
[1708627673.680082][2751:2753] CHIP:TOO: Endpoint: 5 Cluster: 0x0000_0402 Attribute 0x0000_0002 DataVersion: 1432097269
[1708627673.680104][2751:2753] CHIP:TOO:   MaxMeasuredValue: 32766
[1708627673.680161][2751:2753] CHIP:TOO: Endpoint: 5 Cluster: 0x0000_0402 Attribute 0x0000_FFFC DataVersion: 1432097269
[1708627673.680189][2751:2753] CHIP:TOO:   FeatureMap: 0
[1708627673.680577][2751:2753] CHIP:EM: <<< [E:40266i S:27924 M:193327468 (Ack:110210122)] (S) Msg TX to 1:0000000000000001 [8C1C] [UDP:[fe80::e65f:1ff:fe87:d5bf%eth0]:5540] --- Type 0001:01 (IM:StatusResponse)
[1708627673.687551][2751:2753] CHIP:EM: >>> [E:40266i S:27924 M:110210123 (Ack:193327468)] (S) Msg RX from 1:0000000000000001 [8C1C] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1708627673.687586][2751:2753] CHIP:EM: Found matching exchange: 40266i, Delegate: 0xffff7400a100
[1708627673.687622][2751:2753] CHIP:EM: Rxd Ack; Removing MessageCounter:193327468 from Retrans Table on exchange 40266i
[1708627679.911042][2751:2753] CHIP:DMG: Time out! failed to receive report data from Exchange: 40266i
[1708627679.911142][2751:2753] CHIP:DMG: MoveToState ReadClient[0xffff7400a0f0]: Moving to [      Idle]
[1708627679.911220][2751:2753] CHIP:TOO: Error: src/app/ReadClient.cpp:638: CHIP Error 0x00000032: Timeout
[1708627679.911894][2751:2751] CHIP:CTL: Shutting down the commissioner
[1708627679.912110][2751:2751] CHIP:CTL: Shutting down the controller
[1708627679.912176][2751:2751] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1708627679.912230][2751:2751] CHIP:IN: SecureSession[0xffff7400ae70]: MarkForEviction Type:2 LSID:27924
[1708627679.912284][2751:2751] CHIP:SC: SecureSession[0xffff7400ae70, LSID:27924]: State change 'kActive' --> 'kPendingEviction'
[1708627679.912415][2751:2751] CHIP:IN: SecureSession[0xffff7400ae70]: Released - Type:2 LSID:27924
...
[1708627679.916244][2751:2751] CHIP:DL: System Layer shutdown
[1708627679.917016][2751:2751] CHIP:TOO: Run command failure: src/app/ReadClient.cpp:638: CHIP Error 0x00000032: Timeout
```

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

Whatever the Matter 1.2 TH uses

### Platform

raspi, core

### Anything else?

It always fails at the same attribute.

I also tried to pair Apple Homekit to the bridge app (accepting it as an uncertified device). Homekit also tries a wildcard subscription and it fails the same way. Therefore, Homekit immediately re-subscribes and so on.

Running `./chip-tool any read-by-id 0xFFFFFFFF 0xFFFFFFFF 1 0xFFFF` instead results in "CHIP:TOO: Run command failure: IM Error 0x00000501: General error: 0x01 (FAILURE)".

Running the above steps using e.g. chip-all-clusters-app instead in step 2 makes step 4 succeed.

Christoph6848, 2025-02-11:
Any progress on this topic?? We currently met the same problem. Especially the 

./chip-tool any read-by-id 0xFFFFFFFF 0xFFFFFFFF 1 0xFFFF instead results in "CHIP:TOO: Run command failure: IM Error 0x00000501: General error: 0x01 (FAILURE)".

Blocker for TC-IDM-11.1. We also plan to do certification on 1.2.

woody-apple, 2025-04-07:
Closing this, as clearly this is passing cert now for the general case. Please re-open if this is still an issue.

closed 2025-04-07 by woody-apple