project-chip/connectedhomeip #32274: [EFR32] Explain why adv_min/adv_max durations have to be different

tcarmelveilleux, 2024-02-22:
The docs for `sl_bt_advertiser_set_timing` say:

> - Note: interval_max should be bigger than interval_min

It doesn't explain why this should be so, and many examples have copied the requirement all the way up to CHIPDeviceConfig. This can lead to odd choices in trying to get particular rates to be advertised.

Emill, 2024-02-22:
Note the "should" and not "must". From the Bluetooth spec:

"The Advertising_Interval_Min shall be less than or equal to the Advertising_Interval_Max. The Advertising_Interval_Min and Advertising_Interval_Max should not be the same value to enable the Controller to determine the best advertising interval given other activities."

If you have an embedded device running BLE advertising with no other radio activity, there is no issue setting the two values equal.

Also, if you set them differently, it is kind of pointless to set max to min+1. Better give a broader range so the controller has more freedom of choice.

I would assume many "simple" BLE controllers simply pick the min or the max interval, or the average. Compared to scan and connection events, I don't really see the point of accepting a range for adv intervals since a random addition of up to 10 ms is added to every interval, to avoid air collisions.

jmartinez-silabs, 2024-02-23:
Small context;
The Term `should` on the Silabs BLE documentation comes from [RFC-2119](https://www.rfc-editor.org/rfc/rfc2119) and is in the style of Bluetooth core spec.
It is defined as follows:

>    SHOULD   This word, or the adjective "RECOMMENDED", mean that there
>    may exist valid reasons in particular circumstances to ignore a
>    particular item, but the full implications must be understood and
>    carefully weighed before choosing a different course.

More specifically about the BLE advertising min and max interval constraint the BLE core spec says the following:

> The Connection_Interval_Min parameter shall not be greater than the Connection_Interval_Max parameter.

I discussed with the Silabs BLE team and they agree that there isn't any issue with providing the same value for min and max intervals into our API at Silabs level nor the BLE spec level. 
They also agree that the Silabs `sl_bt_advertiser_set_timing` API documentation must be re-worded along the lines of:

> interval_max **must** be greater than or equal to interval_min
or the same wording presented above from the BLE core spec.

In Conclusion, we can use the same Min and Max advertisement interval value in Silabs `BLEManagerImpl` and Silabs BLE doc will be updated.

closed 2024-02-23 by jmartinez-silabs