project-chip/connectedhomeip #32292: [1.2] threadnetworkdiagnostics active-network-faults-list not according to spec

olavt, 2024-02-23:
### Reproduction steps

Issue this chip-tool command:

./chip-tool threadnetworkdiagnostics read active-network-faults-list 1 0

This can result in:

[1708709580.875154][6970:6972] CHIP:TOO:   ActiveNetworkFaultsList: 4 entries
[1708709580.875242][6970:6972] CHIP:TOO:     [1]: 0
[1708709580.875295][6970:6972] CHIP:TOO:     [2]: 0
[1708709580.875348][6970:6972] CHIP:TOO:     [3]: 0
[1708709580.875402][6970:6972] CHIP:TOO:     [4]: 0

The specification states:

"The ActiveNetworkFaults attribute SHALL indicate the set of faults currently detected by the Node.
When the Node detects a fault has been raised, the appropriate NetworkFaultEnum value SHALL be
added to this list. This list SHALL NOT contain more than one instance of a specific NetworkFaultEnum value. When the Node detects that all conditions contributing to a fault has been cleared, the
corresponding NetworkFaultEnum value SHALL be removed from this list. An empty list SHALL
indicate there are currently no active faults."

The current behaviour conflicts with:

"This list SHALL NOT contain more than one instance of a specific NetworkFaultEnum value"

The value 0 does not indicate no error according to the specification, but "Indicates an unspecified fault."

I beleive that in this case it should have been reported an empty list rather than a list with 4 unspecified faults.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

latest

### Platform

core

### Type

Spec Compliance Issue

bzbarsky-apple, 2024-02-28:
@olavt What is the server in this case?

bzbarsky-apple, 2024-02-28:
Ah, I guess WriteThreadNetworkDiagnosticAttributeToTlv always does this?

PR #32391 (2024-03-01): [ThreadDiag]Encode an empty list as no ActiveNetworkFault are tracked
fixes #32292
Attribute value was not respecting the spec.  Encode an empty list as there is no active fault (none are tracked)

Result:
```
[1709244937.426632][5801:5803] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0035 Attribute 0x0000_003E DataVersion: 953547807
[1709244937.426760][5801:5803] CHIP:TOO:   ActiveNetworkFaultsList: 0 entries
```
closed 2024-03-01 by mergify[bot]