project-chip/connectedhomeip #32340: [BUG] chip-bridge-app does not implement an action with id 0xA001, as required by TC-ACT-3.2

Emill, 2024-02-27:
### Reproduction steps

TC-ACT-3.2 specifies in the test plan pdf file that the TH has an action with id 0xA001. As "expected outcome", verify that the received ActionID containts 0xA001.

But the bridge app does not have an action with this id. Instead it has 0x1001 (4097). The Test_TC_ACT_3_2.yaml file instructs the tester to use 4097 rather than 0xA001.

This was tested on RPi Matter 1.2 Test Harness.

### Bug prevalence

N/A

### GitHub hash of the SDK that was being used

Whatever the Matter 1.2 TH uses

### Platform

core

### Anything else?

Please fix so we have consistency about the action id to use for the test. Either change the id in the pdf spec to 4097 or modify the bridge app software, yaml file and the excel test plan verification file to use 0xA001 instead.

bzbarsky-apple, 2024-02-28:
This sounds like an issue for https://github.com/project-chip/matter-test-scripts ...

Emill, 2024-02-28:
@bzbarsky-apple should I repost there issue there instead?

bzbarsky-apple, 2024-03-01:
@Emill I think that would be a good idea, yes.

KishokG, 2024-08-20:
Action with id 0xA001 is not implemented in chip-bridge-app, as required by TC-ACT-3.2 test plan.

When I trying to run instant-action command with id 0xA001, I'm getting `status = 0x8b (NOT_FOUND),`response.