project-chip/connectedhomeip #32353: [Darwin] MTRDevice should detect parts list change and inform delegate

jtung-apple, 2024-02-28:
As title says. Part of this is to investigate what are all the necessary / sufficient signals for this, and how it should be conveyed.

bzbarsky-apple, 2024-02-28:
Some thoughts on what might be relevant here:

1. PartsList changes.
2. AcceptedCommandList changes.
3. AttributeList changes.
4. DeviceTypeList changes.
5. ClusterRevision changes.
6. FeatureMap changes.

nivi-apple, 2024-04-24:
Following are the changes that were tracked.
PartsList changes.
AcceptedCommandList changes -> Cluster revision should be updated for this.
AttributeList changes  -> Cluster revision should be updated for this.
DeviceTypeList changes.
ClusterRevision changes.
FeatureMap changes.
ServerList is needed for changes in clusters that are supported per endpoint

PR #33146 (2024-05-03): Add support for notifying the delegate when device configuration changes
- Device configuration changes include updates in attributes parts list, server list, device type list, cluster revision or feature map in the descriptor cluster

- Add a test to verify the delegate is notified when device configuration changes

Fixes: #32353
closed 2024-05-03 by mergify[bot]