project-chip/connectedhomeip #32397: [1.3] How to update the Attribute value of Electrical Power Measurement Cluster and Electrical Energy Measurement Cluster?

Semi-Bin-Blain, 2024-03-04:
### Reproduction steps

I enabled the Electrical Power Measurement Cluster(Server ) and Electrical Energy Measurement(Server) Cluster in my application， but I can't update the Attribute value, and I can not read the attribute from my device by the chip-tool, chip-tool returns `FAILURE`;
In the file [zzz_generated/app-common/app-common/zap-generated/attributes/Accessors.cpp](https://github.com/project-chip/connectedhomeip/blob/9f6d627e0262e1d023986291948bb4e845be803e/zzz_generated/app-common/app-common/zap-generated/attributes/Accessors.cpp), for these two clusters, there is only the ClusterRevision attribute;
Then I attempted to use the function `emberAfWriteAttribute`, but it returned `FAILURE`.
Am I using it in the right way or not? If I am using it incorrectly, could you please provide me with the proper usage?
Thank you in advance for all your responses.
Below is the code:
```Cpp
Clusters::ElectricalEnergyMeasurement::Structs::MeasurementAccuracyStruct::Type measurementAccuracy;
Clusters::ElectricalEnergyMeasurement::Structs::MeasurementAccuracyRangeStruct::Type measurementAccuracyRange;
measurementAccuracy.measured = true;
measurementAccuracy.measurementType = Clusters::detail::MeasurementTypeEnum::kVoltage;
measurementAccuracy.maxMeasuredValue = 65535;
measurementAccuracy.minMeasuredValue = 0;

measurementAccuracyRange.rangeMax = 187647121162243;
measurementAccuracyRange.rangeMin = 4294967295;
measurementAccuracyRange.percentMax.SetValue(50698);
measurementAccuracyRange.percentMin.SetValue(0);
measurementAccuracyRange.percentTypical.SetValue(0);
measurementAccuracyRange.fixedMax.SetValue(187650443764698);
measurementAccuracyRange.fixedMin.SetValue(187650443764696);
measurementAccuracyRange.fixedTypical.SetValue(0);
measurementAccuracy.accuracyRanges = {measurementAccuracyRange};

status = emberAfWriteAttribute(kExampleEndpointId, Clusters::ElectricalEnergyMeasurement::Id, Clusters::ElectricalEnergyMeasurement::Attributes::Accuracy::Id, (uint8_t*)&measurementAccuracyRange, ZCL_STRUCT_ATTRIBUTE_TYPE);
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

9f6d627e0262e1d023986291948bb4e845be803e

### Platform

other

### Type

Common Cluster Logic

bzbarsky-apple, 2024-03-04:
@Semi-Bin-Blain These clusters ask a delegate for the values.  See https://github.com/project-chip/connectedhomeip/blob/9f6d627e0262e1d023986291948bb4e845be803e/src/app/clusters/electrical-power-measurement-server/electrical-power-measurement-server.h#L35 for example.  You implement that interface, pass your implementation to the cluster instance object you create (see https://github.com/project-chip/connectedhomeip/blob/9f6d627e0262e1d023986291948bb4e845be803e/src/app/clusters/electrical-power-measurement-server/electrical-power-measurement-server.h#L104-L110), and the functions on your interface will be called when the cluster needs an attribute value.

Note that it's your responsibility to call `MatterReportingAttributeChangeCallback` when those attribute values change.

closed 2024-03-04 by bzbarsky-apple

jamesharrow, 2024-03-05:
You may also find it helpful to look at the https://github.com/project-chip/connectedhomeip/blob/master/examples/energy-management-app/energy-management-common/src/ElectricalPowerMeasurementDelegate.cpp as an example

Semi-Bin-Blain, 2024-03-12:
Thanks to all of you! @jamesharrow @bzbarsky-apple

aravindone, 2025-05-06:
Hello @bzbarsky-apple  and @jamesharrow  as i tried to do the same, i have tried to updated value using set function  and read value through get functions, but the updated value is not reflected in chip repl. below is my code 
`

static ElectricalPowerMeasurementDelegate gPowerDelegate;
void SimulateMeasurements()
{
    ChipLogProgress(AppServer, "Inside simulate measurement");

    int64_t voltage = 230;  // Example: 230V
    int64_t power   = 50;   // Example: 50W
    int64_t current = 10;   // Example: 10A

    // Set attributes
    gPowerDelegate.SetVoltage(DataModel::Nullable<int64_t>(voltage));
    gPowerDelegate.SetActivePower(DataModel::Nullable<int64_t>(power));
    gPowerDelegate.SetActiveCurrent(DataModel::Nullable<int64_t>(current));

    // Get attributes back and print
    auto readVoltage = gPowerDelegate.GetVoltage();
    auto readPower   = gPowerDelegate.GetActivePower();
    auto readCurrent = gPowerDelegate.GetActiveCurrent();

    if (!readVoltage.IsNull())
    {
        ChipLogProgress(AppServer, "Voltage after setting: %lld V", readVoltage.Value());
    }
    else
    {
        ChipLogError(AppServer, "Voltage is NULL after setting!");
    }

    if (!readPower.IsNull())
    {
        ChipLogProgress(AppServer, "Power after setting: %lld W", readPower.Value());
    }
    else
    {
        ChipLogError(AppServer, "Power is NULL after setting!");
    }

    if (!readCurrent.IsNull())
    {
        ChipLogProgress(AppServer, "Current after setting: %lld A", readCurrent.Value());
    }
    else
    {
        ChipLogError(AppServer, "Current is NULL after setting!");
    }
}`

output image:
[image]

looking forward for your feedback