project-chip/connectedhomeip #32411: [Documentation] Clarification/Advice for Cluster Data version and Login on "Reboots"/"Restarts"and Firmware-Updates

Apollon77, 2024-03-04:
### Documentation issues

When I read the specs the cluster data version is set per cluster instance on first initialization (or after factory reset) and then increased per changed value. So far so easy 🙂

Question A: When a device restarts/reboots (also after a Firmware-Update potentially) all "defaults" are applied for all non-volatile attributes ... but because of the fact that most values are not persisted we do not know which defaults changed potentially by a "firmware update" or the initial device state changes (e.g. because temperature is now different). This would that mean that cluster data version for all clusters should be increased once to make sure that controllers like Apple, that use a held data version on their side and work with cluster data version filtern for subscriptions, really get the newest state updated for their subscription. Or?

Question B: When the answer to A is "yes" then why it makes sense to store the cluster data version at all? We could simply randomly initialize clusters versions anew on each startup - that would gain the same effect (as by all is updated) when we make sure the new value is different from old one. And for most i/o limited "in field devices" this would also reduce i/o because always storing the cluster version could lead to early storage issues (considering storages used with limited write cycles). Real question then is where the benefit is to "just" increase on reboot/restart of the device?

QuestionC: If A is "wrong/no" then this would mean we need to store all data (or a checksum or such) in order to decide if any of the volatile/default values got changed to just increase the version in that case ... that would mean way more storage space needed ... but that would make sure to just increase the version when needed.

I wonder how this is interpreted specification wise?

Thank you for a clarification or advice whats the "correct" (and practical) way.

Apollon77, 2024-03-04:
Maybe @bzbarsky-apple or @woody-apple can give their opinions from Apple pov because Apple uses (as only controller right now from what I saw so far) a held data version ...

bzbarsky-apple, 2024-03-05:
> When a device restarts/reboots

In the SDK, this resets all data versions to random values.

In the spec, it says:

> A cluster data version SHALL be initialized randomly when it is first published.

but it does not really clearly define what "first published" means.

> why it makes sense to store the cluster data version at all?

On the server?  It doesn't, and they are not stored in the SDK; they are kept in RAM only.

closed 2024-03-05 by bzbarsky-apple

Apollon77, 2024-03-06:
Thank you also here for your advice. I think it makes sense to clarify this in the specification then what "first published" measn and that it is ok that it is not persisted. Thank you