project-chip/connectedhomeip #32433: [1.2] chip-tool temperaturecontrol set-temperature command

salvio96, 2024-03-05:
### Reproduction steps

After the pairing, I'm trying to use the following commands to trying to set the temperature level

- chip-tool temperaturecontrol set-temperature 1 1 --TargetTemperature 1
- chip-tool temperaturecontrol set-temperature 1 1 --TargetTemperatureLevel 1

The first one doesn´t work, while the second works correctly.

- CHIP:TOO: Run command failure: IM Error 0x00000585: General error: 0x85 (INVALID_COMMAND)
- CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0056 Command=0x0000_0000 Status=0x0   and the statusIB is Succes 

I tried to undestrand the reason and it seems that the frature map is always 0x2
- CHIP:TOO:   FeatureMap: 2

In the source code the enum of the feature map is
- 0x1 for TargetTemperature
- 0x2 for TargetTemperatureLevel

So the first command in not already supported at all, or I'm missing something?

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

181b0cb14ff007ec912f2ba6627e05dfb066c008

### Platform

other

### Platform Version(s)

Linux

### Type

Platform Issue

bzbarsky-apple, 2024-03-06:
Sounds like the specific server you are running against doesn't support TargetTemperature.  @salvio96 what is the issue, exactly?

salvio96, 2024-03-07:
When I try to use this command

chip-tool temperaturecontrol set-temperature 1 1 --TargetTemperature 1

**Output app**

[1709799694.227264][11149:11149] CHIP:DMG: Endpoint= 1, Cluster= 0x00000056  nameSpace= TemperatureControl
[1709799694.227276][11149:11149] CHIP:DMG: Cluster= 0x00000056  SUPPORTED
[1709799694.227326][11149:11149] CHIP:DMG: Received command for Endpoint=1 Cluster=0x0000_0056 Command=0x0000_0000
[1709799694.227344][11149:11149] CHIP:DMG: Endpoint=1 Cluster=0x0000_0056 Command=0x0000_0000 status 0x85 (INVALID_COMMAND) (no additional context)

**Output chi-tool**

[1709799694.227637][253528:253530] CHIP:DMG: ICR moving to [ResponseRe]
[1709799694.227664][253528:253530] CHIP:DMG: InvokeResponseMessage =
[1709799694.227672][253528:253530] CHIP:DMG: {
[1709799694.227679][253528:253530] CHIP:DMG: 	suppressResponse = false, 
[1709799694.227687][253528:253530] CHIP:DMG: 	InvokeResponseIBs =
[1709799694.227699][253528:253530] CHIP:DMG: 	[
[1709799694.227705][253528:253530] CHIP:DMG: 		InvokeResponseIB =
[1709799694.227715][253528:253530] CHIP:DMG: 		{
[1709799694.227721][253528:253530] CHIP:DMG: 			CommandStatusIB =
[1709799694.227734][253528:253530] CHIP:DMG: 			{
[1709799694.227740][253528:253530] CHIP:DMG: 				CommandPathIB =
[1709799694.227750][253528:253530] CHIP:DMG: 				{
[1709799694.227757][253528:253530] CHIP:DMG: 					EndpointId = 0x1,
[1709799694.227765][253528:253530] CHIP:DMG: 					ClusterId = 0x56,
[1709799694.227773][253528:253530] CHIP:DMG: 					CommandId = 0x0,
[1709799694.227782][253528:253530] CHIP:DMG: 				},
[1709799694.227793][253528:253530] CHIP:DMG: 				
[1709799694.227800][253528:253530] CHIP:DMG: 				StatusIB =
[1709799694.227809][253528:253530] CHIP:DMG: 				{
[1709799694.227817][253528:253530] CHIP:DMG: 					status = 0x85 (INVALID_COMMAND),
[1709799694.227830][253528:253530] CHIP:DMG: 				},
[1709799694.227836][253528:253530] CHIP:DMG: 				
[1709799694.227842][253528:253530] CHIP:DMG: 			},
[1709799694.227852][253528:253530] CHIP:DMG: 			
[1709799694.227858][253528:253530] CHIP:DMG: 		},
[1709799694.227868][253528:253530] CHIP:DMG: 		
[1709799694.227874][253528:253530] CHIP:DMG: 	],
[1709799694.227885][253528:253530] CHIP:DMG: 	
[1709799694.227891][253528:253530] CHIP:DMG: 	InteractionModelRevision = 11
[1709799694.227900][253528:253530] CHIP:DMG: },
[1709799694.227931][253528:253530] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0056 Command=0x0000_0000 Status=0x85
[1709799694.227944][253528:253530] CHIP:TOO: Error: IM Error 0x00000585: General error: 0x85 (INVALID_COMMAND)

While using command 

./chip-tool temperaturecontrol set-temperature 1 1 --TargetTemperatureLevel 2

**Output app**

[1709799437.638685][11149:11149] CHIP:EM: Handling via exchange: 52377r, Delegate: 0x5878c0
[1709799437.638750][11149:11149] CHIP:DMG: Endpoint= 1, Cluster= 0x00000056  nameSpace= TemperatureControl
[1709799437.638753][11149:11149] CHIP:DMG: Cluster= 0x00000056  SUPPORTED
[1709799437.638767][11149:11149] CHIP:DMG: Received command for Endpoint=1 Cluster=0x0000_0056 Command=0x0000_0000
[1709799437.638772][11149:11149] CHIP:ZCL: Size temperature control 
[1709799437.638774][11149:11149] CHIP:DMG: Endpoint 1, Cluster 0x0000_0056 update version to 920613cd

**Output chip-tool**

[1709799437.639036][252335:252337] CHIP:DMG: ICR moving to [ResponseRe]
[1709799437.639046][252335:252337] CHIP:DMG: InvokeResponseMessage =
[1709799437.639049][252335:252337] CHIP:DMG: {
[1709799437.639051][252335:252337] CHIP:DMG: 	suppressResponse = false, 
[1709799437.639054][252335:252337] CHIP:DMG: 	InvokeResponseIBs =
[1709799437.639060][252335:252337] CHIP:DMG: 	[
[1709799437.639062][252335:252337] CHIP:DMG: 		InvokeResponseIB =
[1709799437.639067][252335:252337] CHIP:DMG: 		{
[1709799437.639069][252335:252337] CHIP:DMG: 			CommandStatusIB =
[1709799437.639073][252335:252337] CHIP:DMG: 			{
[1709799437.639075][252335:252337] CHIP:DMG: 				CommandPathIB =
[1709799437.639079][252335:252337] CHIP:DMG: 				{
[1709799437.639081][252335:252337] CHIP:DMG: 					EndpointId = 0x1,
[1709799437.639084][252335:252337] CHIP:DMG: 					ClusterId = 0x56,
[1709799437.639087][252335:252337] CHIP:DMG: 					CommandId = 0x0,
[1709799437.639089][252335:252337] CHIP:DMG: 				},
[1709799437.639093][252335:252337] CHIP:DMG: 				
[1709799437.639095][252335:252337] CHIP:DMG: 				StatusIB =
[1709799437.639099][252335:252337] CHIP:DMG: 				{
[1709799437.639102][252335:252337] CHIP:DMG: 					status = 0x00 (SUCCESS),
[1709799437.639104][252335:252337] CHIP:DMG: 				},
[1709799437.639107][252335:252337] CHIP:DMG: 				
[1709799437.639109][252335:252337] CHIP:DMG: 			},
[1709799437.639112][252335:252337] CHIP:DMG: 			
[1709799437.639114][252335:252337] CHIP:DMG: 		},
[1709799437.639117][252335:252337] CHIP:DMG: 		
[1709799437.639122][252335:252337] CHIP:DMG: 	],
[1709799437.639126][252335:252337] CHIP:DMG: 	
[1709799437.639128][252335:252337] CHIP:DMG: 	InteractionModelRevision = 11
[1709799437.639130][252335:252337] CHIP:DMG: },
[1709799437.639145][252335:252337] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0056 Command=0x0000_0000 Status=0x0

The Zap configuration is in the following screen

[image]
[image]
[image]

bzbarsky-apple, 2024-03-07:
OK, so the app configuration is that the TL (TemperatureLevel) feature is enabled and the TN (TemperatureNumber) and STEP (TemperatureStep) features are not enabled (see FeatureMap attribute).

Per spec, the TargetTemperature field is supposed to be present exactly when when the TN feature is enabled.  Likewise, the TargetTemperatureLevel field is supposed to be present exactly when the TL feature is enabled.

So the server correctly responds with INVALID_COMMAND when it gets a command with TargetTemperature, since the server does not support TN.

Again @salvio96, what is the issue?  What behavior were you expecting and why?

bzbarsky-apple, 2024-03-07:
And to be clear: _this specific server application_ is configured to support TL and not TN.  You could have a different configuration, where you support TN and not TL.  Per spec you have to support exactly one of those two.