project-chip/connectedhomeip #32449: "sed -i" doesn't work on mac, it should be "sed -i ''" instead

gaoyangsandy, 2024-03-05:
This line doesn't work on mac.
https://github.com/project-chip/connectedhomeip/blob/6b026afebe68270df60058e95afa152ca30781fd/examples/chef/chef.py#L700

it should be -i '' instead per https://thoughtbot.com/blog/sed-102-replace-in-place

gaoyangsandy, 2024-03-05:
Here's the failed message when I run it directly.

sed -i 's/CONFIG_DISABLE_IPV4=y/#\\ CONFIG_DISABLE_IPV4\\ is\\ not\\ set/g' sdkconfig
sed: 1: "sdkconfig": unterminated substitute pattern

sxb427, 2025-04-02:
This is now fixed - 

Darwin and linux specific variants - https://github.com/project-chip/connectedhomeip/blob/master/examples/chef/chef.py#L733-L738

Darwin: `sed -i ''`
Linux: `sed -i `

andy31415, 2025-04-02:
Closing as fixed in #33529

closed 2025-04-02 by andy31415