project-chip/connectedhomeip #32451: [Test Failed] TotalOperationalHours is Cleared after Reset

Yinxq, 2024-03-06:
### Test issue(s)

1. perform factory reset
2. ./chip-tool generaldiagnostics read total-operational-hours 1 0
3. Wait for 1 hour 5 minutes
4. ./chip-tool generaldiagnostics read total-operational-hours 1 0
5. Reboot target device(DUT)
6. ./chip-tool generaldiagnostics read total-operational-hours 1 0

In step 4, totaloperationalhours is 1 and in step 6 it is 0, but according to the test plan, the value should still be 1.

Besides, I would like to ask a question. If I press reset after running for 1h50min, and then press reset after running again for 1h50min, should totaloperationalhours be 2 or 3? The spec doesn't specify this in detail.
...

### Platform

nrf connect

Damian-Nordic, 2024-03-07:
I'm not sure if the test case is valid given that according to the spec "The TotalOperationalHours attribute SHALL indicate a **best-effort** attempt at tracking the length of time".

Overall, those "best-effort" features in Matter are very hard to verify. In this case, the challenge to implement this accurately is that storing the total operational hours too frequently into persistent storage would needlessly wear the flash. Given this and that the attribute has rather low significance for the user, we decided to save the attribute into flash every 10 hours (this is configurable with `CONFIG_CHIP_OPERATIONAL_TIME_SAVE_INTERVAL` Kconfig option).

I can see that other platforms don't do even that and only save this value when the platform is gracefully shutdown, so if I understand correctly, hard reset will lose the number of operational hours since the last boot.

Yinxq, 2024-03-11:
Hi @Damian-Nordic. Thanks so much for your quick reply! I confirmed again that this is an optional test, and your approach is much more appropriate. So I will close this issue.

closed 2024-03-11 by Yinxq