project-chip/connectedhomeip #32452: [BUG] report from server side "The Resource is busy and cannot process the request" during CASE session establishing process.

nykh2010, 2024-03-06:
### Reproduction steps

Pairing node with ble-wifi. chip-tool start a establishment, send sigma1 message 4 times, but miss response. Seconds later, chip-tool restart a new establishment, the node report "Resource busy".
Does the CASE Server has a timeout to re-init the CASE establish process? I could not found.

### Bug prevalence

low probability

### GitHub hash of the SDK that was being used

181b0cb14ff007ec912f2ba6627e05dfb066c008

### Platform

esp32, core

### Platform Version(s)

IDF version 4.4.5, Ubuntu version 22.04

### Anything else?

[chip-tool.log](https://github.com/project-chip/connectedhomeip/files/14504372/chip-tool.log)
[node.log](https://github.com/project-chip/connectedhomeip/files/14504375/node.log)

bzbarsky-apple, 2024-03-06:
> Does the CASE Server has a timeout to re-init the CASE establish process?

Yes, it will time out eventually when it does not get a response to the Sigma2 it sent.

nykh2010, 2024-03-07:
> > Does the CASE Server has a timeout to re-init the CASE establish process?
> 
> Yes, it will time out eventually when it does not get a response to the Sigma2 it sent.

In this case, is it the client-side retry a new CASE connection before server-side timeout cause the "Resource busy" fault ?
I found the code, for client-side, response timeout is 2s, for server-side, timeout is 30s. Is that right? why not set the server-side timeout shorter than client-side? so the server can correctly receive the new request from the client.

bzbarsky-apple, 2024-03-13:
> In this case, is it the client-side retry a new CASE connection before server-side timeout cause the "Resource busy" fault ?

Yes.

>  found the code, for client-side, response timeout is 2s, for server-side, timeout is 30s. Is that right?

Yes, because processing a Sigma1 message is normally cheap but processing a Sigma2 message requires a lot of cryptographic computation.

Note that when a client does retries, it in fact tries to give more time than just the timeout on its own send before the retry.  And if it gets a BUSY it uses (after some recent changes) the value provided in the BUSY response as the retry delay.

mideayanghui, 2024-05-14:
@bzbarsky-apple Hi Boris, is this issue will be fixed  ? Now here is the same problem using matter code while start a CASE session. (tag: v1.2.0.1) 
[chiptool_log.txt](https://github.com/project-chip/connectedhomeip/files/15307561/chiptool_log.txt)
[dut.log](https://github.com/project-chip/connectedhomeip/files/15307578/dut.log)

bzbarsky-apple, 2024-05-15:
> Is this issue will be fixed ?

What issue, exactly?  The server is waiting for a response to its Sigma2.  It can only handle one CASE handshake at a time.  It tells the client to try again later.  Later it stops waiting; if the client retries, it should succeed at that point.