project-chip/connectedhomeip #32462: [1.0] RuntimeError: Please install the official package with: pip install grpcio

dafatefa888, 2024-03-06:
### Reproduction steps

when i get the source code of matter SDK v1.0-branch and run the` source scripts/activate.sh`，it occured that i had not install the grpcio which is already installed in my ubuntu:
[image]
here is the error:

`Requirement already satisfied: grpcio>=1.0.0 in /home/test/matter/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages (from googleapis-common-protos->pigweed==0.0.10+20240306195657) (1.62.0)
  Using cached googleapis-common-protos-1.3.0.tar.gz (26 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting grpc>=1.0.0 (from googleapis-common-protos->pigweed==0.0.10+20240306195657)
  Using cached grpc-1.0.0.tar.gz (5.2 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-3qs1qnku/grpc_842b4100ad4e49339695caf934ab2a80/setup.py", line 33, in <module>
          raise RuntimeError(HINT)
      RuntimeError: Please install the official package with: pip install grpcio
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

ninja: build stopped: subcommand failed.
['ninja', '-C', '/home/test/matter/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']

Traceback (most recent call last):
  File "/home/test/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 327, in install_packages
    subprocess.check_call(ninja_cmd, stdout=outs, stderr=outs)
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['ninja', '-C', '/home/test/matter/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.`

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

latest version in 1.0 branch: 9bd53e85d91d996d6dfc645ca71be5a43830f1b4

### Platform

other

### Type

Platform Issue

sanlingye, 2024-03-15:
Did you manage to resolve this problem?

dafatefa888, 2024-03-18:
> I'm also seeing the same error. Did you manage to resolve this?

not yet，just switch to tag v1.0.0.1

dafatefa888, 2024-03-18:
> Did you manage to resolve this problem?

no

whatcall, 2024-03-30:
save issue. Switch to tag v1.0.0.2 since tag v1.0.0.1 is not available yet.