project-chip/connectedhomeip #32478: [Feature] [chip-tool] send web socket on interactive server mode

mmt-10, 2024-03-07:
### Feature description

Hi,

I would like chip-tool to send the information of the subscribe cluster it received via web socket.

Execute `./chip-tool interactive server`.
Set `onoff subscribe on-off 5 10 <node id> 1` via web socket.

chip-tool shows received data from cluster's subscribed message.
However, chip-tool did not send received message via web socket.

### Platform

core

PokersKun, 2024-03-25:
Also looking at this, any solutions? @vivien-apple @bzbarsky-apple

bzbarsky-apple, 2024-03-26:
Someone would have to write code for this.

Chapoly1305, 2024-03-26:
It's quite complex to add feature into the project itself, but I think it might be possible to create a workaround. The chiptool can use trace parameter to generate a live json file, maybe we can setup a separated ws server to publish periodic messages until officially supported. Maybe a Python script to consume the content of the json file?

PhLuReh, 2024-07-03:
I think current implementation is more or less not thought to be asynchronous (which would in fact be supported by websocket). But the implementation allows for an empty (or numeric only for specifying a timeout in s) message, which will be used as [async call](https://github.com/project-chip/connectedhomeip/blob/4ef104b1019c05469483c7391d549a0407302846/examples/chip-tool/commands/interactive/InteractiveCommands.cpp#L326). So if results are reported (for example by the registered subscriptions), those will be returned by the next read after async call.

This is my exemplary Python implementation, which is hopefully helpful:
``` python
async def recv(websocket):
    while True:
        # send async call
        await websocket.send("")
        # wait for response
        msg = await websocket.recv()
        now = datetime.now()
        default = True
        try:
            data = json.loads(msg)
            # if result was json and contains results
            if "results" in data:
                for res in data["results"]:
                    # if it was a change of an attribute value. print it
                    if "endpointId" in res:
                        print(f"[{now}] endpoint {res['endpointId']}: {res['attributeId']}={res['value']}")
                        default = False
        except Exception:
            pass
        if default:
            print(f"{msg}")
```

Having a timeout set in the async call also enables to receive logs and failures. But I also resulted in some Segmentation Fault over time (10-20s).

mmt-10, 2024-07-19:
@PhLuReh

Thank you for your solution.
Our member tried your solution.

We make 2 websocket interfaces as below;
for Sending command (input only) ws://localhost:9002
for Subscribe command (listen only) ws://localhost:9003

Then, Segmentation Fault disappeared.