project-chip/connectedhomeip #32480: [Build] Could not compile the ChipTool in Xcode 15

muralivm, 2024-03-07:
### Build issue(s)

1. Download the Darwin code
2. Open the Darwin.workspace
3. Select the chiptool scheme 
4. Set signing the provision profiles
5. Run the project

Run script build phase 'Run GN Build (chip_xcode_build_connector)' will be run during every build because the option to run the script phase "Based on dependency analysis" is unchecked. is throwing error.

...

### Platform

darwin

### Anything else?

Showing All Issues
Run script build phase 'Run GN Build (chip_xcode_build_connector)' will be run during every build because the option to run the script phase "Based on dependency analysis" is unchecked.

PhaseScriptExecution Run\ GN\ Build\ (chip_xcode_build_connector) /Users/mm2/Library/Developer/Xcode/DerivedData/Darwin-chpywfiwbmbvteaqbmnrhkvsxhmk/Build/Intermediates.noindex/Matter.build/Debug-iphonesimulator/Matter.build/Script-0C40A67D246C9AC700885C81.sh (in target 'Matter' from project 'Matter')
    cd /Users/mm2/Library/CloudStorage/OneDrive-Allegion/Learning\ materials/My\ Repo/connectedhomeip-master/src/darwin/Framework
    export ACTION\=build
    export AD_HOC_CODE_SIGNING_ALLOWED\=YES
    export ALLOW_TARGET_PLATFORM_SPECIALIZATION\=NO
    export ALTERNATE_GROUP\=staff
    export ALTERNATE_MODE\=u+w,go-w,a+rX
    export ALTERNATE_OWNER\=mm2
    export ALWAYS_EMBED_SWIFT_STANDARD_LIBRARIES\=NO
    export ALWAYS_SEARCH_USER_PATHS\=NO
    export ALWAYS_USE_SEPARATE_HEADERMAPS\=NO
    export APPLE_INTERNAL_DEVELOPER_DIR\=/AppleInternal/Developer
    export APPLE_INTERNAL_DIR\=/AppleInternal
    export APPLE_INTERNAL_DOCUMENTATION_DIR\=/AppleInternal/Documentation
    export APPLE_INTERNAL_LIBRARY_DIR\=/AppleInternal/Library
    export APPLE_INTERNAL_TOOLS\=/AppleInternal/Developer/Tools
    export APPLICATION_EXTENSION_API_ONLY\=NO
    export APPLY_RULES_IN_COPY_FILES\=NO
    export APPLY_RULES_IN_COPY_HEADERS\=NO
    export APP_SHORTCUTS_ENABLE_FLEXIBLE_MATCHING\=YES
    export ARCHS\=x86_64
    export ARCHS_STANDARD\=arm64\ x86_64
    export ARCHS_STANDARD_32_64_BIT\=arm64\ i386\ x86_64
    export ARCHS_STANDARD_32_BIT\=i386
    export ARCHS_STANDARD_64_BIT\=arm64\ x86_64
    export ARCHS_STANDARD_INCLUDING_64_BIT\=arm64\ x86_64
    export ARCHS_UNIVERSAL_IPHONE_OS\=arm64\ i386\ x86_64
    export ASSETCATALOG_COMPILER_GENERATE_ASSET_SYMBOLS\=YES
    export ASSETCATALOG_FILTER_FOR_DEVICE_MODEL\=iPhone15,4
    export ASSETCATALOG_FILTER_FOR_DEVICE_OS_VERSION\=17.0.1
    export ASSETCATALOG_FILTER_FOR_THINNING_DEVICE_CONFIGURATION\=iPhone15,4
    export AUTOMATICALLY_MERGE_DEPENDENCIES\=NO
    export AVAILABLE_PLATFORMS\=appletvos\ appletvsimulator\ driverkit\ iphoneos\ iphonesimulator\ macosx\ watchos\ watchsimulator
    export AppIdentifierPrefix\=9KX5HL49NQ.
    export BITCODE_GENERATION_MODE\=marker
    export BUILD_ACTIVE_RESOURCES_ONLY\=YES
    export BUILD_COMPONENTS\=headers\ build
    export BUILD_DIR\=/Users/mm2/Library/Developer/Xcode/DerivedData/Darwin-chpywfiwbmbvteaqbmnrhkvsxhmk/Build/Products
    export BUILD_LIBRARY_FOR_DISTRIBUTION\=NO
    export BUILD_ROOT\=/Users/mm2/Library/Developer/Xcode/DerivedData/Darwin-chpywfiwbmbvteaqbmnrhkvsxhmk/Build/Products
    export BUILD_STYLE\=
    export BUILD_VARIANTS\=normal
    export BUILT_PRODUCTS_DIR\=/Users/mm2/Library/Developer/Xcode/DerivedData/Darwin-chpywfiwbmbvteaqbmnrhkvsxhmk/Build/Products/Debug-iphonesimulator
    export BUNDLE_CONTENTS_FOLDER_PATH_deep\=Contents/
    export BUNDLE_EXECUTABLE_FOLDER_NAME_deep\=MacOS
    export BUNDLE_EXTENSIONS_FOLDER_PATH\=Extensions
    export BUNDLE_FORMAT\=shallow
    export BUNDLE_FRAMEWORKS_FOLDER_PATH\=Frameworks
    export BUNDLE_PLUGINS_FOLDER_PATH\=PlugIns
[... 662 lines cut ...]
Checking out: nlassert, nlio, nlunit-test, mbedtls, qrcode, pigweed, openthread, nanopb, third_party/pybind11/repo, third_party/jsoncpp/repo, editline, third_party/boringssl/repo/src, third_party/libwebsockets/repo, perfetto

fatal: not a git repository (or any of the parent directories): .git

Traceback (most recent call last):

  File "/Users/mm2/Library/CloudStorage/OneDrive-Allegion/Learning materials/My Repo/connectedhomeip-master/scripts/checkout_submodules.py", line 148, in <module>

    main()

  File "/Users/mm2/Library/CloudStorage/OneDrive-Allegion/Learning materials/My Repo/connectedhomeip-master/scripts/checkout_submodules.py", line 141, in main

    checkout_modules(selected_modules, args.shallow, args.force, args.recursive)

  File "/Users/mm2/Library/CloudStorage/OneDrive-Allegion/Learning materials/My Repo/connectedhomeip-master/scripts/checkout_submodules.py", line 104, in checkout_modules

    subprocess.check_call(cmd)

  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call

    raise CalledProcessError(retcode, cmd)

subprocess.CalledProcessError: Command '['git', '-C', '/Users/mm2/Library/CloudStorage/OneDrive-Allegion/Learning materials/My Repo/connectedhomeip-master', 'submodule', '--quiet', 'update', '--init', '--depth', '1', 'third_party/nlassert/repo', 'third_party/nlio/repo', 'third_party/nlunit-test/repo', 'third_party/mbedtls/repo', 'examples/common/QRCode/repo', 'third_party/pigweed/repo', 'third_party/openthread/repo', 'third_party/nanopb/repo', 'third_party/pybind11/repo', 'third_party/jsoncpp/repo', 'third_party/editline/repo', 'third_party/boringssl/repo/src', 'third_party/libwebsockets/repo', 'third_party/perfetto/repo']' returned non-zero exit status 128.

Command PhaseScriptExecution failed with a nonzero exit code

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/

muralivm, 2024-03-15:
> Matter Commission demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/

Thanks for your response, are you able to compile Darwin workspace ?

JoeyYang2010, 2024-08-29:
@muralivm Same issue here, did you find any solution?Thanks.

JoeyYang2010, 2024-09-05:
@muralivm If you download the source code, it will fail, if you git clone it, everything will be ok.

Alami-Amine, 2024-12-16:
closing since @JoeyYang2010  provided an answer no feedback from OP

closed 2024-12-16 by Alami-Amine